OpenBuilds/OpenBuilds-CONTROL #12: GRBL reporting option

swarfer, 2018-09-08:
After using this driver and then going back to bCNC I was not seeing any position reports.
Looks like this driver set $10=2 while the defauilt is $10=1

According to the GRBL wiki $10=2 is not recommended for GUI's
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#10---status-report-mask

If you are wanting BUF reports then the value should be $10=3  

(I tried to just make a Wiki page to inform users but cannot.)

openbuilds-engineer, 2018-09-14:
Isn't $10=3 an old Grbl 0.9 thing, no longer in Grbl 1.1? https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#10---status-report-maskbinary

openbuilds-engineer, 2018-09-14:
Fixed the parser to understand MPOS, WPOS and BUF (So doesnt matter whether $10 is 0,1 or 2) - Travis/AppVeyor will compile it into v.1.0.92 - when you see it available for download, please help me test (: - make sure it shows workpos for whatever config correctly 
if $10 is 0 or 2, then we get WPOS, just displays WPOS in the DRO.
if $10=1 then we get MPOS.  Subtract WCO from MPOS to get WPOS and display result in DRO

closed 2018-09-14 by openbuilds-engineer

openbuilds-engineer, 2018-09-14:
https://github.com/OpenBuilds/SW-Machine-Drivers/releases/tag/v1.0.92 (or newer)

swarfer, 2018-09-19:
I tested with $10=3 and got the right numbers so far.  Will report back if anything strange happens in the future.

commit cdc14ef7 (2018-09-14): v1.0.92 - fix Parser to handle MPOS and WPOS to fix #12