OpenBuilds/OpenBuilds-CONTROL #17: OpenBuilds Machine Driver never quits (Mac OS)

zirkeltraining, 2018-09-30:
(with OB MD c.1.0.95 on Mac OS 10.13.6 - haven't tested it elsewhere)

Closing the window with the "x" Button or choosing "Quit" from the file menu does not end the application. Instead it remains active in the dock at all times. Must use "force quit" to get rid of it, which feels ... dirty.
This should not be the default behaviour. Also, there seems to be no config option to avoid the app to start upon boot. 
That's also not what a nice driver should do, isn't it?
Please include an option to easily kill the driver and to avoid it coming back up on restart.

openbuilds-engineer, 2018-10-01:
There is a Quit

[image]

Regarding the other behavior:
1)  The idea here is that it sits in the background similar to a "printer driver" (Think HP, Epson etc) - but due to cross platform nature its not entirely as transparent
2) Being always-on is needed for the integration with cam.openbuilds.com and upcoming projects 
3) This is pretty standard industry behaviour (see Cricut's agent, EaselDriver, etc) -   Doing it this way helps those users who are not IT professionals - they dont need to worry, it just works. They dont need to remember to run something first, or start it or whatever, its just always there. 

I'll consider adding an option to disable autostart - will keep issue tagged and close once implemented.

openbuilds-engineer, 2018-10-01:
I'll also checkout https://stackoverflow.com/questions/35008347/electron-close-w-x-vs-right-click-dock-and-quit to catch the Mac native "Quit" menu (Mac user count is much lower than windows, so its OS specific quirks arent all handled yet)

openbuilds-engineer, 2018-11-05:
I've increased the forcefulness of "quitting" in the app, but my OSX VM is broken (no Mac to test on)
Version 1.0.112 will be available later today (compiling now) - if you want to give it some testing and please report back. 

Regarding opting out of the autostarting, still waiting for a fix in https://github.com/electron/electron/issues/10880 -  the problem is still upstream at the Electron project - once thats fixed, only then can I do something on our side (: - thanks for the patience

zirkeltraining, 2018-11-14:
I appreciate your thorough response, here's what I find on the mac with 1.0.112:

- File->Quit or apple-Q or right-clicking the dock icon and using "Quit" still only closes the drivers' window.
- but right-clicking the dock icon and using "Quit Machine driver" seems to work fine (as before; I only hadn't noticed this menu)
- the dock context menu "options->open at login" is not checked when the menu is opened for the first time after a machine boot, although the driver was autostarted. That's probably just an initialisation missing. After opening the menu for the second time, it's checked.
- however, the checkmark on "open at login" cannot be removed - it remains on all the time.
- during autostart, apparently the driver window opens with no controls at first, then takes multiple seconds to become responsive and redrawn.

summary: 
- there are two quit options in the dock menu, but only one does what i'd expect
- autostart can never be turned off at the moment. must quit the driver manually after each boot to get rid of it.

openbuilds-engineer, 2018-11-15:
Re "open at login" and removing from autostart, remember its upstream from us, see https://github.com/electron/electron/issues/10880 - has since been fixed in https://github.com/electron/electron/pull/15010 - unfortunately they merged it into Electron 4 (We're not on 4 yet) so its not a quick fix to add it now, we need to do the whole upgrade so its something that'll have to wait till after the busy season i'm afraid.  

You found the right menu though so thats the way for now.  

Mac stuff is hard, as I dont have one to test one );

openbuilds-engineer, 2018-11-15:
Going to close this issue as the fixes will happen when we move to Electron 4 automatically

closed 2018-11-15 by openbuilds-engineer