OpenBuilds/OpenBuilds-CONTROL #20: Keyboard Shortcuts

sharmstr, 2018-10-25:
Just a suggestion to add keyboard shortcuts for a few things like start, pause and abort.

sharmstr, 2018-11-26:
Start = Alt + R
Pause = Spacebar
Stop = Esc
Jog x = left and right arrows
Jog y = up and down arrows
jog z = page up and page down

openbuilds-engineer, 2019-01-16:
Thoughts on Gamepad vs Keyboard?
(Gamepads have a more standardised layout, makes easier to learn)

sharmstr, 2019-01-16:
Are the gamepads basically just keyboard emulators?  You map a button on the gamepad to a key (or combination of keys)?

openbuilds-engineer, 2019-01-16:
Not entirely: see https://w3c.github.io/gamepad/#remapping - its a standardised input (so there are canonical standards as to what the events report, so without having to need a complex mapping procedure, the "standard" api will know your up Dpad, is my Up Dpad, etc.   The point you are heading too (Just make it all customizable) is great, but doesnt align with the goals of this project. You'll notice there is no "settings" button in CONTROL for a reason, it should 100% be plug and play, no setup required.  (Grbl Settings doesnt count because we have to setup the firmware, but still, no setup to the application itself needed) 

Whats nice about Gamepad vs Keyboard is that I can take exclusive control of a gamepad (and read inputs whether the window is in focus or not) but with Keyboard, you need to be aware of other applications listening for the same events, and the window has to be "in focus" or active before the events get fired.

openbuilds-engineer, 2019-01-16:
Perhaps I should spin up a demo to play with to convince you guys (;

sharmstr, 2019-01-16:
Thank you for the explanation.  It sounds like the gamepad fits your vision and they are cheap enough.

openbuilds-engineer, 2019-01-22:
```
var curvalue = 0; // set externally from websocket
var direction = X- // X-, X, Y-, Y, Z-, Z // set externally from websocket

setInterval(function() {
  if (curvalue > 10) {
    if (!blocked) {
      console.log("$J=G91 G21 " + direction + "1 F" + curvalue * 4 + "\n")
      port.write("$J=G91 G21 " + direction + "1 F" + curvalue * 4 + "\n");
      blocked = true;
    }
  } else {
    port.write(String.fromCharCode(0x85));
    // console.log("STOP ")
  }
}, 20);
```

Dug up the old code from +-8 months ago - got reminded of it by @swarfer in the forum, remembered I wrote it already.  Posting here so I dont forget where it is again

mobilemcclintic, 2019-02-20:
In staying "plug and play" with the software, I'd like to see a customizable binding menu, but default it to standardish settings.  I'm just me, but I would be thinking along the lines of the below.  Personally the best feature that keeps me coming back to Open Builds control is the troubleshooting tab and the axis steps/mm adjustment function.

Start/Play = Default to nothing, bindable (I don't have an idea)
Pause = Default to nothing, bindable (I don't have an idea)
Stop/Abort = Default to nothing, bindable
Soft E-Stop = Default to nothing, bindable
Jog (X) = Left Joystick, keypad 2/8
Jog (Y) = Left Joystick, keypad 4/6
Jog (Z) = Right Joystick, keypad 3/9
Jog Size+ = Right trigger button / keypad "-"
Jog Size- = Left trigger button / keypad "+"
Feed Override+10% = controller directional pad Up
Feed Override-10% = controller directional pad Down
Spindle Override+10% = controller directional pad Left
Spindle Override-10% = controller directional pad Right
Setzero X = Default to nothing, bindable (Nice to have, but if not intuitive could mess things up)
Setzero Y = Default to nothing, bindable (Nice to have, but if not intuitive could mess things up)
Setzero Z = Default to nothing, bindable (Nice to have, but if not intuitive could mess things up)
Setzero XYZ =  Default to nothing, bindable (Nice to have, but if not intuitive could mess things up)

stevemayathome, 2019-03-11:
I'm Partial to seeing left arrow -X right arrow +X, up arrow +Y down arrow -Y
with the page up +Z page down -Z
But
How do I hookup a game pad to my pc, this sounds good too.
just my nickel's worth (we don't have penny's anymore)

jbhurst, 2019-04-12:
I think you can stick with "plug and play" goal and still offer some customization for advanced users without forcing them to fork.  One way would be to define standard shortcuts, but have them editable in a JSON, but not readily exposed through menus.  Since so many games do this, including games in Python, there is probably some open binding code you could pillage.  Sorry, I don't know of any off hand though.

Having custom bindings also would help a lot to enable customers that want a custom controller.  I've thought about buying one of the programmable game or CAD/CAM controllers with OLED screens and configuring it to run a CNC and execute certain standard macros.

openbuilds-engineer, 2019-04-16:
[image]

(; this might just be in 1.0.153 or later (;

stevemayathome, 2019-04-16:
That looks like a great start, only question is when. Please don't keep us hanging. I like the customizability aspect.

openbuilds-engineer, 2019-04-18:
v1.0.156 is out! 

Bonus! Also now supports Continuous Jogging (Hold down mouse or keyboard button to jog. Longer you hold, further it goes!. Lift off button and it stops (with deceleration)   -  consider this part of the feature a little beta for now

[image]

And then of course: here's the Keyboard setup (some defaults preconfigured)
[image]

Closing for now!

closed 2019-04-18 by openbuilds-engineer

stevemayathome, 2019-04-18:
Just downloaded and tried out. What took you so long (trying to be funny)
only problem I had was with my anti virus program. It wanted to send it away for analysis. I said yes but let it go right now. So AVG should have it on file now. It works great and is intuitive. Cute the way you flash the lights when moving. Well done !! Thanks

openbuilds-engineer, 2019-04-18:
Sorry to hear about your AV. 
We have bought some Microsoft Autheticode Certificates will be adding that to the build process soon.  Not to hijack this issue, but does your AV show why it reacts that way?  (New issue please)

stevemayathome, 2019-04-18:
hi;
I'm running win10 with latest updates, the antivirus is AVG with it's latest updates. 

I did let my Windows Defender Firewall know to let the program go through. 

Typically when I load a new program (or as well as any previous releases of OpenBuilds Control) the AV will pop up and do a 10 - 15 second scan of the program. 

This time it did the 10 - 15 second search then it came back with a " we haven't seen this before can we send it to them so that they could look at it more deeply for about 52 seconds. ( why 52 I don't know)

It wouldn't allow me to say no and install it anyway. 

So I cancelled the install, opened the AV control panel and changed configuration so that I control what is allowed. ( sounds like the old Norton stuff I used to fight with ) 

I then restarted the install and after the first 10 -15 second check it came up and asked again but also had the option to go ahead anyway, as well as sending them the file to look at.

 I expect that they want to check it and allow it in future checks. That's just a guess on my part.

Anything else I can help with just ask.

Thx Steve

openbuilds-engineer, 2019-04-18:
Ahh ok. So just them being a little paranoid.oh well. Stopped using AVG
when they went all spammy a couple years ago. Windows Defender does the job
for me

jbhurst, 2019-04-19:
This is great!  Thanks!

sharmstr, 2020-10-30:
Can you add feed and tool override as well?