OpenBuilds/OpenBuilds-CONTROL #30: Cant see path

docteurfrankenstine, 2018-12-03:
Hi,
Everything was working good but 1 month ago after an update i cant see any more the path, i see it on cam.

openbuilds-engineer, 2018-12-05:
I'll need a little more info:
1) Screenshot of your Settings tab in CAM
2) Export the GCODE, ZIP and attach to this issue please
3) How are you loading GCODE into OpenBuilds Control?  (From the Open button, or using the API in cam.openbuilds.com at the bottom right with the Transfer GCODE button?)

openbuilds-engineer, 2018-12-05:
Also: 4.   Do you have a 3D viewer tab in OpenBuilds CONTROL/MachineDriver?  (If not, check the Serial Log when loading the window - it may print something like WebGL not working on your machine)

docteurfrankenstine, 2018-12-07:
Hello,
i try to send code with the green button bottom right also to save code and to open it in open builds control, same result… i also update my graphics card driver..
best regards.

openbuilds-engineer, 2018-12-07:
Please include the exported gcode as a zipped attachment, and the
screenshot etc as requested before. Cannot help you if i dont have those.

openbuilds-engineer, 2018-12-07:
Please answer/upload required information for points 1-4 above... Cannot troubleshoot without the information (complete). Thanks

openbuilds-engineer, 2018-12-19:
User does not respond to debug information, cannot assist without. Closing issue - if you someday revisit, upload the requested information and answer the questions, then we can help you (:

closed 2018-12-19 by openbuilds-engineer