OpenBuilds/OpenBuilds-CONTROL #43: Upgrade doesnt stop process so upgrade fails. Windows10

sharmstr, 2019-01-14:
The upgrade modal states that it will stop the process but it doesnt.  It merely closes the window.  Because of this, the software never updates.  It will stay on the current version even though you go through the update install.  There are no failed install warnings.

The workaround is to click "ok" to start the upgrade.  Once the window closes and before going through the upgrade, manually close the process by right clicking the icon in the sys tray.  Once the driver is stopped, proceed with the upgrade like normal.

openbuilds-engineer, 2019-01-14:
I use the same online update procedure as the users, and have not had that happen to me yet - and I make sure to test each version before clicking "release" too.  Is there maybe another windows PC nearby you can check to eliminate that particular machine?  - the closing happens during the install (After launching the installer, the installer detects "OpenbuildsCONTRON is running, do you want to close it" or something to that wording

openbuilds-engineer, 2019-01-14:
[image]

sharmstr, 2019-01-14:
I've never seen that "openbuildscontrol is running" message.

openbuilds-engineer, 2019-01-15:
Try a different pc? We have three win10 laptops in the shop that work fine, i have Win7 here on the testbed, also fine.

You can manually grab an older build from
https://github.com/OpenBuilds/SW-Machine-Drivers/releases

Install say 1.0.129 first
Then download 1.0.131 and run it while 1.0.129 is running. Should popup. The installer does that, not our application, so you dont need to use the Online update - running the installer in any way will detect it.

Another thing you can try: uninstall completely, install an older version, then try a new one?

sharmstr, 2019-01-17:
Didnt have an issue upgrading last night to 1.0.133.  Still didnt get the "ok to close it" message, but it did upgrade.  Go ahead an close this.

closed 2019-01-17 by sharmstr