OpenBuilds/OpenBuilds-CONTROL #44: F360 Integration

openbuilds-engineer, 2019-01-17:
Similar to https://github.com/tapnair/UGS_Fusion

sharmstr, 2019-03-05:
Awesome idea.  Not that you're asking me, but I'd rather have to post do it then a plugin :)

openbuilds-engineer, 2019-05-29:
Coming soon! (What you see may still change...)

[image]

[image]

[image]

petervanderwalt, 2020-06-22:
Abandoning plugin https://docs.openbuilds.com/doku.php?id=docs:software:fusion360

closed 2020-06-22 by petervanderwalt