OpenBuilds/OpenBuilds-CONTROL #49: "Check Size" button doesn't recognize G20 (inch mode) in file

tbmartin211, 2019-01-27:
If I load a GCode file written in inches (G20) - the "Check Size" button still uses millimeters. I have to manually send a G20, then hit the "Check Size" button for it to work properly.  

Probably a second issue - Also noticed when I send the manual G20, the jog buttons use inches (button labels don't change)! Yikes...when you are expecting 10mm...and it moves 10 inches!  I would prefer the jog buttons to keep using mm, but not sure the right answer here - G21X10...?  Handling mixed units can get complicated...

I've been working around it with G20 - check size, G21 - jog to zero, (sometimes repeat), run GCode file.

Overall, love the driver...simple and works.

openbuilds-engineer, 2019-01-29:
Good catch, its on the todo now (:

Officially its Metric so bare with me adding proper inch mode support will take a while

openbuilds-engineer, 2019-01-29:
Not done yet (: sorry the commit closed the issue: Fixed the Jog buttons... still need to fix Check Size. Stand by

closed 2019-01-29 by openbuilds-engineer

openbuilds-engineer, 2019-01-29:
https://github.com/OpenBuilds/SW-Machine-Drivers/releases/tag/v1.0.140 and up... Please test (: