OpenBuilds/OpenBuilds-CONTROL #51: MACROs still has a lot of growth ahead (multiline macros, scripting, variables etc) -

openbuilds-engineer, 2019-01-30:
MACROs still has a lot of growth ahead (multiline macros, scripting, variables etc) -
@swarfer from https://openbuilds.com/threads/openbuilds-control-software.13121/page-7#post-76203

jbhurst, 2019-05-29:
I've been looking for a way to re-order macros and can't find it.  Can it be done via a text file somewhere?  Once you have a few already set up, if you want to add a new one (say related to the first 3 and needing to group with them), I can't figure out how to do that without deleting them all and starting over.  The ability to use our own icons (.ico, .bmp, whatever) would be nice too, even if not exposed in the UI. 

PS - thanks for all the recent changes.  I'm moving back to Control from bCNC for normal operations and only using it for macros/operations that need a camera.

sharmstr, 2019-09-06:
https://openbuilds.com/threads/openbuilds-control-software.13121/page-29#post-86666

MikonosII, 2020-03-23:
How about hot key support for macros?

petervanderwalt, 2020-04-14:
> How about hot key support for macros?

Checkout v1.0.223 when it comes out in a day or two 

[image]

[image]

petervanderwalt, 2020-04-17:
> MACROs still has a lot of growth ahead (multiline macros, scripting, variables etc) -
> @swarfer from https://openbuilds.com/threads/openbuilds-control-software.13121/page-7#post-76203

Full Javascript execution (: coming to you in 1.0.229 or later  - access to all the variables, all the functions, and all the magic of JS (setTimeouts, calculations, etc)  

[image]

[image]

closed 2020-04-17 by petervanderwalt