OpenBuilds/OpenBuilds-CONTROL #57: Probe wizard for X / Y axis?

lhilton, 2019-03-14:
Is there any effort underway or planned to implement an X / Y probe wizard? I use a probing system that lets me find the edges as well as surface and having that edge finding portion in the wizard would speed up my work piece setup time.

If there is no complaint about adding this, I can do the dev work and provide a pull request.

sharmstr, 2019-03-20:
Workaround is to add a macro.  Works great.

openbuilds-engineer, 2019-03-25:
At this time we only supply a Z Touch-off plate https://openbuildspartstore.com/z-touch-plate/ so the software, officially, only has a Z touch off wizard to match the ecosystem 
As mentioned above, for custom tasks, we have the Macros

closed 2019-03-25 by openbuilds-engineer