OpenBuilds/OpenBuilds-CONTROL #61: Feature Request - help text for macros

jbhurst, 2019-04-12:
Would love a field in macro setup to enter descriptive text that will pop up when a user hovers over a macro allowing them to see more about what it does.  I'm setting up a machine for operator use and want them to pick from a few predefined actions, but it is hard to describe with an icon and two or three words.

openbuilds-engineer, 2019-04-12:
Cool idea, will add todo

openbuilds-engineer, 2019-04-12:
V1.0.151 (out in couple days) will have this:

[image]
[image]

closed 2019-04-12 by openbuilds-engineer