OpenBuilds/OpenBuilds-CONTROL #63: Program hangs on loss of communication to CNC

jbhurst, 2019-04-12:
My testing is using a xPro V3 driven over USB from Windows 10 on a LEAD1010 setup.  

Several times now, it has gotten into a state where it no longer response to mouse or keyboard input and a single CPU core is maxed until the program is force terminated.

It seems to only happen after the machine has some sort of problem and has to be E-Stopped - typically a motor driver problem that results in the slave Y axis stopping mid movement.  After that, the OpenBuilds Controller sometimes (but I don't think always) locks up as described.  Is this a known problem?  If I get more details on replication, I'll update.

openbuilds-engineer, 2019-04-12:
First report but we do handle disconnects, check the serial log.  It doesnt lock up for me

openbuilds-engineer, 2019-04-12:
PS: Forum first this is more of a dev to-do, use the forum for discussion. Once proven as a bug we'll take it to git

closed 2019-04-12 by openbuilds-engineer

openbuilds-engineer, 2019-04-15:
Improved in v 1.0.151 or later by the way