OpenBuilds/OpenBuilds-CONTROL #67: Display job duration/remaining time

sharmstr, 2019-04-18:
https://openbuilds.com/threads/openbuilds-control-software.13121/page-14#post-80639

openbuilds-engineer, 2019-04-18:
Now I got to go and remember the history.  It worked at some point, then didnt... from the old LW days some of this code came. 

We do calc the time for each segment still...  Will check later

https://github.com/OpenBuilds/SW-Machine-Drivers/blob/master/app/lib/3dview/workers/gcodeparser.js#L817-L836

openbuilds-engineer, 2019-04-18:
[image]

Well, good news is I have the sum of all the moves stores on the userData for each line segment.  So we can do something simple:

total time = object.userData.lines[object.userData.lines.length-1].p2.timeMinsSum
remaining time = total time - (runtime so far or from  object.userData.lines[numberoflinesalreadysent].p2.timeMinsSum)

openbuilds-engineer, 2019-04-18:
Concept checks out:

[image]
Will add it as overlay on the bottom progress bar in next release (and formatted nicer)

sharmstr, 2019-04-18:
Awesome!

openbuilds-engineer, 2019-04-18:
[image]

v1.0.156 or later (:

closed 2019-04-18 by openbuilds-engineer