OpenBuilds/OpenBuilds-CONTROL #73: Keyboard Jog

openbuilds-engineer, 2019-04-26:
If you

Hold jog, get close to where you wanna be, let off, and then double tap, it goes into a neverending jog

openbuilds-engineer, 2019-04-30:
WIP: https://codepen.io/petervanderwalt/pen/BEEJPy

batcrave, 2020-03-03:
I repro'd it a little differently - holding jog, then a rapid release-press-release - and can't get the double-tap version to work at all, but the part that really puzzled me is that I could repro it on the X & Z axis, but *not* on Y - and  I've only briefly skimmed keyboard.js, but the code for all three axes looks identical.

Then, after switching to one of the stock machine configs, it would happen on all three axes. 

Trying to recreate my previous settings, I cranked up Y accel ($121) way up and... no bug. At 500mm/s/s I can still trigger it maybe one in five tries. By 600mm/s/s I'm not quick enough to trigger it at all (at least with the keyboard - I haven't tested it using the flaky mouse or a professional Starcraft player)

The fact that the click/tap speed required to set it off varies with acceleration leads me to suspect either:
A) The problem's somewhere up or downstream of keyboard.js, and the grbl end of things may be playing a part in the problem, or else
B) There's something I fundamentally misunderstood about how keyboard.js is working.

(yeah, probably B)

In any case, it's not a solution - or even a particularly useful workaround - but it's another data point.

petervanderwalt, 2020-04-16:
So, with https://github.com/OpenBuilds/OpenBuilds-CONTROL/commit/fc224ddc327a7606573b1db01261040e86f33b4c I changed the behaviour of the buttons - would be worth keeping an eye if it didn't perhaps by coincidence make a difference to the stuck key bug. Here's to hoping

petervanderwalt, 2020-11-16:
Implemented checking loop in 1.0.254-1.0.257
Change in behaviour:
If you are using Continuous jogging, we wait until Grbl is in "Idle" state before sending jogs. Too many jogs in queue messed with the Stopping via the cancel jog command. 
If you attempt a jog with Grbl not in Idle state, you get a little toast error and log entry: "Please wait for machine to be Idle, before jogging. Try again once it is Idle"

Also added soft-limits checking, also relying on being Idle: -  Toast/Log will now tell you Unable to jog toward " + axis + ", will hit soft-limit if it generates a move shorter than 1mm (ie will hit soft limit if we continue)

closed 2020-11-16 by petervanderwalt

petervanderwalt, 2020-11-28:
Should now be fully resolved in 1.0.262