OpenBuilds/OpenBuilds-CONTROL #77: taskbar icon size, Linux

swarfer, 2019-05-09:
On a Linux machine the openbuilds icon on the taskbar fills the entire screen.
[image]
in the image the battery icon is normally at far right of the screen, but now the black and white OB 'icon' takes up the entire width, pushing the other icons to the left.

This is probably related to the icon file being of very large pixel count as reported in another issue.  Fixing that may fix this, but my 2 different machines display different symptoms.  on one the icon in the taskbar is large, in the other the icon on the menu is large.

openbuilds-engineer, 2019-05-10:
XFCE  (: What year is this! Lol

Electron takes a standard icon as input and rescales it during the built as appropriate. Don't know if they sypport XFCE
LXDE, Gnome and Unity doesnt do this for what its worth.  

Will see if I can do anything, but suspect its going to need fighting upstream

swarfer, 2019-05-10:
hehe, don't like Mate, so I install Xubuntu

the icon file is about 2000x2000 pixels so I suspect that if you supply Electron with a 32x32 everyone will be happy.

thing is that on one machine the menu did not scale it to fit and on the other the taskbar did not scale it to fit.

openbuilds-engineer, 2019-05-10:
https://www.electron.build/icons we use the first option right now: Linux icon set will be generated automatically based on the macOS icns file or common icon.png.

But yeah, can change to use the second option: 
Or you can put them into the build/icons directory if you want to specify them yourself. The filename must contain the size (e.g. 32x32.png) of the icon). Recommended sizes: 16, 24, 32, 48, 64, 96, 128, 256 (or just 512).

swarfer, 2019-05-10:
the current icon.png is 1024x1024 but I am certain I saw more than 2000x2000 in GIMP last night.

and the 32x32.png file is 207kb and 2089x1980 and that must be what I saw on my machines.
surely it should be an actual 32x32 image?

openbuilds-engineer, 2019-05-10:
Last Build Log shows https://travis-ci.org/OpenBuilds/OpenBuilds-CONTROL/jobs/529970808#L831 
- IE electron-builder did the heavy lifting. 

Will play someday when I am tasked on Software again (Hardware mode from Mon again) and manually create the files, specify electron-builder to use them instead of recreate them from the ICNS file.

swarfer, 2019-05-10:
ok, so long as there is a plan..

openbuilds-engineer, 2019-06-04:
As is so often the case with open source libraries, the docs is scarce, examples few, and even following what they have to a tee, it still grabs the icons from the icons.icns file, not from the individual pngs when building under TravisCI. Just an update that I did take a little look.  Will keep at it!

petervanderwalt, 2020-11-16:
Is this still happening after all the changes over the last couple months

closed 2020-12-28 by petervanderwalt