OpenBuilds/OpenBuilds-CONTROL #78: Additional key binding options

sharmstr, 2019-05-15:
https://openbuilds.com/threads/openbuilds-control-software.13121/page-18#post-82219

Individual Zero for X, Y and Z
Switch for Continuous/incremental jogging.
+- is perfect, in continious it can change feed, in incremental change stepsize

openbuilds-engineer, 2019-05-15:
Thanks for adding it (: 
Adding to note for myself too... but good chance to investigate how to add Macros here too...

WMFB, 2019-05-15:
Thanks for logging the request.

petervanderwalt, 2020-04-14:
```
Switch for Continuous/incremental jogging.
```

[image]
Added, refer https://github.com/OpenBuilds/OpenBuilds-CONTROL/issues/121 

```
Bonus: if you are in continuous mode, and hit +/- it jumps back to Incremental too (just felt natural in the workflow - as you are used to +/- affecting Incremental functionality, once you go for those, its good that it goes to incremental mode then (:
```

Still keeping issue open for individual XYZ buttons someday (:

petervanderwalt, 2020-04-14:
> Adding to note for myself too... but good chance to investigate how to add Macros here too...

Did that ...  v1.0.223

[image]
[image]

WMFB, 2020-04-14:
> > Adding to note for myself too... but good chance to investigate how to add Macros here too...
> 
> Did that ... v1.0.223
> 
> [image]
> [image]

:) Happy Clap.... THANK YOU...

closed 2020-04-14 by petervanderwalt