OpenBuilds/OpenBuilds-CONTROL #79: X and Y probing (Community demands)

openbuilds-engineer, 2019-05-17:
Probably one of the most popular requests in regards to probing is to add wizard for X and Y too  (And full XYZ)

ssbxfire, 2019-05-21:
It would be good to have XY probing for both outside edge and inside edge, Triple Edge Finders is used for inside location.

openbuilds-engineer, 2019-05-21:
@ssbxfire 

One thing to note, is that we have to be very careful, and sometimes seem inflexible, to community requests. 

As at the end of the day we are a business, there are many factors that shape the directive, apart from popular demand.  
One of the biggest factors is EASE OF USE.  OpenBuilds CONTROL is an EASY TO USE, BEGINNER FRIENDLY tool to get people brand new to CNC, started.  In a way I hope users outgrow it as that means they've grown their skills and needs! The goal is not to become the ULTIMATE Grbl host (: so things like the various probing options you suggest, will just add to the complication for new users, and at the same time add a support burden on the support team as users will get stuck trying to do that. 

What makes it in is very much based on "We get asked about this a lot - what can we do to make it easier" 

(: So thanks for the suggestions, but do keep that in mind, the goal here is driven by the penultimate banner: EASY!

openbuilds-engineer, 2019-05-21:
Oh, and for those who want to stay on CONTROL even once they've outgrown it (; we did give you Macros (; to do what whatever you want (:

ssbxfire, 2019-05-21:
I fully understand, I'm just putting my .02 on a request already entered

cpapplefamily, 2019-10-06:
It seem this software is just that. "Too simple" maybe following the path of other great open source maker company that supply here software. Having a easy/advanced mode button would be beneficial.

kasperv79, 2020-01-07:
Hi. 
Is the xy Homing  functionally something that is beingworked on or?? 
And when is it planned to be released?

petervanderwalt, 2020-02-06:
When is hard to say, it is being worked on, and the software side is not the delay (clue hehe)

petervanderwalt, 2020-04-14:
Implemented in v1.0.221 or later

closed 2020-04-14 by petervanderwalt