OpenBuilds/OpenBuilds-CONTROL #88: f you switch from Incremental to Continuous, or vice versa, all numpad direction moves are locked out until you click on one of the six control buttons with the mouse.

openbuilds-engineer, 2019-06-12:
f you switch from Incremental to Continuous, or vice versa, all numpad direction moves are locked out until you click on one of the six control buttons with the mouse.

tbirdglenn, 2019-06-13:
I just found out the same condition holds true if you change the jog feed rate in mm/min. The keypad is locked out until you click a direction with the mouse.

tbirdglenn, 2019-06-13:
After further review, I believe the second comment above is not a bug. While the cursor is focused in the mm/min box, the keypad is locked out and that is a good thing. You do not have to move the machine to unlock the keypad, just click in an open area to take the cursor focus out of the box.

The same is true for the first comment, you don't have to move the machine, but I believe the keypad should not be locked out just because you switched from incremental to continuous jog.

openbuilds-engineer, 2019-06-13:
Correct all Text fields are excluded from keybindings (as keyboard events
in there is meant to work with the text (move cursor, type, edit etc) so
yeah you need to defocus a text field before keybindings work.

First one is a bug, did test. Fix next week or so, not working on software
at the moment (scheduled time slots)

tbirdglenn, 2019-07-24:
I just installed ver 1.0.180 and switching from Incremental to Continuous, or vice versa, still locks out the keypad until you click somewhere to defocus.

openbuilds-engineer, 2019-07-25:
I just tested again and it works here...

What OS (win10), Browser (Chrome), any other specifics to your setup that differs from what we test on? 100% sure you are on 180 (or 182 - havent made changes to that bits but still good to be on the latest)

closed 2019-07-25 by openbuilds-engineer

tbirdglenn, 2019-07-26:
I hadn’t noticed that 182 was out, I have 180 installed. I have two independent machines. Both have their own BlackBox, both have their own PC running Windows10, neither one is connected to the internet or each other. Both still lock out the keypad until you defocus.
I installed 180 on top of 179. Then tried uninstalling 180, rebooting, and clean installing 180. Problem is still there. If you haven’t made changes to 182, does that mean that this problem would still be there in 182?

tbirdglenn, 2019-07-27:
I just installed 182 and the problem is solved on both machines. Thank you!