OpenBuilds/OpenBuilds-CONTROL #89: Missing a Custom Incremental Jog distance box for the X/Y axis and a separate input box for the Z axis.

tbirdglenn, 2019-06-13:
I'm Falling-In-Like with the OpenBuilds-CONTROL, but in migrating from Universal G Code Sender I miss the two custom jog input boxes, one for X/Y and one for Z. I know we can do a work around by setting an axis to zero, inputting a distance and hitting enter, but that changes the job zero. It's also not repeatable, so if you want to do multiple jogs at a custom distance you have to input the same info each move. The four canned incremental choices are great, but two additional custom input boxes would make it fantastic. 

Also I see in an earlier issue there is a method to prevent the Z axis from responding to the 100mm jog, but I don't see a way to do that.

tbirdglenn, 2019-06-16:
Optional thought for above. If you don't want to add two additional jog input boxes, how about making the 100mm box (or forth box) into a custom input box? Then add an on/off radio button either beside or below the box titled 'Move Z'. The value in that box would be retained for the next power up. Also, the value in that box would convert when switching between mm and inch mode.

openbuilds-engineer, 2019-06-17:
Unfortunately we will not be providing "custom" lengths. It just makes the UI too cluttered.

- Use the Continious Jog or Incremental Jog (keyboard+incremental is sooo easy and dont even have to take eyes off machine) 
- Enter Jog commands in the Serial Terminal https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging
- Use Absolute Positioning by clicking in the DRO numbers

closed 2019-06-17 by openbuilds-engineer