OpenBuilds/OpenBuilds-CONTROL #91: segfault on raspbian buster

hydroflame, 2019-06-26:
related: https://github.com/OpenBuilds/OpenBuilds-CONTROL/issues/86

raspbian buster uses libc6 2.28 by default which causes segfault. OpenBuilds-CONTROL needs exactly 2.27

openbuilds-engineer, 2019-07-08:
https://github.com/electron/electron/issues/16205#issuecomment-508454921 mentions SNAP builds work, havent had time to play with that. Any volunteers? (; 

Will need to add all the Snapcraft dependencies etc, manual here https://www.electron.build/configuration/snap

Something like https://github.com/OpenBuilds/OpenBuilds-CONTROL/wiki/Compiling-on-Raspberry-Pi-%5Bunofficial%5D but at step 6 instead of ```./node_modules/.bin/build --armv7l --linux tar.xz ``` do a SNAP build

petervanderwalt, 2020-03-30:
Closing: Refer https://github.com/OpenBuilds/OpenBuilds-CONTROL/wiki/Run-on-a-Raspberry-Pi-(No-Compile-Method)

closed 2020-03-30 by petervanderwalt