OpenBuilds/OpenBuilds-CONTROL #100: Connection issue with GRBL_ESP32

Geekoid85, 2019-08-17:
I have upgraded an old cnc by replacing the electronic with grbl. It's a version ported to the esp32 and... I have some trouble getting it to connect to openbuilds control.
https://github.com/bdring/Grbl_Esp32
As I seen in the terminal, OBC send a CTRL-X to reset grbl and then hope to see `Grbl 1.1f ['$' for help]`but identify it as a smoothieboard. 

So I'm not sure why as it's working great with cncjs, maybe by sacrificing a usb cable I can inspect the traffic with a logic analyzer and get further analysis. The issue is reproducible I think so if you have an esp32 you can try it by yourself.
https://github.com/bdring/Grbl_Esp32/issues/67
Here is an issue about connecting issue in general.
Thanks !

Geekoid85, 2019-08-18:
I'm wrong, I can directly plug the logic analyzer to the tx and rx gpio of the esp. Before the serial to usb converter.

openbuilds-engineer, 2019-08-20:
I am familiar with ESP32_Grbl but its not really one of our officially supported platforms at this time, sorry, we cannot allocate time to it :( 

As you mentioned: OBC send a CTRL-X to reset grbl and then hope to see Grbl **1.1f
So we arent getting that Grbl 1.1f string back in time as would happen with regular Grbl.  
We wait 1second for the normal arduino autoreset, 
then we wait 2 seconds for Ctrl+X responses
Then if still no valid response, we check for Smoothie

Thats the problem with ports sometimes (not sure if the case here), they deviate too far from what "standard Grbl" does and then they become unsupportable.

closed 2019-08-20 by openbuilds-engineer

openbuilds-engineer, 2019-08-20:
Sidenote: I've got a board on my desk here with an older build of ESP32_grbl:

```
[21:20:37] Detected a change in available ports: [COM4][COM3]
[21:20:42] [ connect ] PORT INFO: Port is now open: COM4 - Attempting to detect Firmware
[21:20:42] [ connect ] Checking for firmware on COM4
[21:20:42] [ connect ] Detecting Firmware: Method 1 (Autoreset)
[21:20:42] [ ] ok
[21:20:43] [ connect ] Detecting Firmware: Method 2 (Ctrl+X)
[21:20:43] [ ] Grbl 1.1f ['$' for help]
[21:20:43] [ [external from hardware] ] OpenBuilds CONTROL received a FEEDHOLD notification from Grbl: This could be due to someone pressing the HOLD button (if connected), or DriverMinder on the xPROv4 detected a driver fault
[21:20:43] [ [external from hardware] ] OpenBuilds CONTROL received a RESET/ABORT notification from Grbl: This could be due to someone pressing the RESET/ABORT button (if connected), or DriverMinder on the xPROv4 detected a driver fault
[21:20:43] [ [external from hardware] ] OpenBuilds CONTROL received a CYCLESTART/RESUME notification from Grbl: This could be due to someone pressing the CYCLESTART/RESUME button (if connected)
[21:20:43] [ $$ ] $0=3 ;Step pulse time, microseconds
[21:20:43] [ $$ ] $1=250 ;Step idle delay, milliseconds
[21:20:43] [ $$ ] $2=0 ;Step pulse invert, mask
[21:20:43] [ $$ ] $3=0 ;Step direction invert, mask
[21:20:43] [ $$ ] $4=0 ;Invert step enable pin, boolean
[21:20:43] [ $$ ] $5=1 ;Invert limit pins, boolean
[21:20:43] [ $$ ] $6=0 ;Invert probe pin, boolean
[21:20:43] [ $$ ] $10=2 ;Status report options, mask
[21:20:43] [ $$ ] $11=0.010 ;Junction deviation, millimeters
[21:20:43] [ $$ ] $12=0.002 ;Arc tolerance, millimeters
[21:20:43] [ $$ ] $13=0 ;Report in inches, boolean
[21:20:43] [ $$ ] $20=0 ;Soft limits enable, boolean
[21:20:43] [ $$ ] $21=0 ;Hard limits enable, boolean
[21:20:43] [ $$ ] $22=0 ;Homing cycle enable, boolean
[21:20:43] [ $$ ] $23=3 ;Homing direction invert, mask
[21:20:43] [ $$ ] $24=200.000 ;Homing locate feed rate, mm/min
[21:20:43] [ $$ ] $25=2000.000 ;Homing search seek rate, mm/min
[21:20:43] [ $$ ] $26=250 ;Homing switch debounce delay, milliseconds
[21:20:43] [ $$ ] $27=1.000 ;Homing switch pull-off distance, millimeters
[21:20:43] [ $$ ] $30=1000.000 ;Maximum spindle speed, RPM
[21:20:43] [ $$ ] $31=0.000 ;Minimum spindle speed, RPM
[21:20:43] [ $$ ] $32=0 ;Laser-mode enable, boolean
[21:20:43] [ $$ ] $100=800.000 ;X-axis steps per millimeter
[21:20:43] [ $$ ] $101=800.000 ;Y-axis steps per millimeter
[21:20:43] [ $$ ] $102=800.000 ;Z-axis steps per millimeter
[21:20:43] [ $$ ] $110=5000.000 ;X-axis maximum rate, mm/min
[21:20:43] [ $$ ] $111=4000.000 ;Y-axis maximum rate, mm/min
[21:20:43] [ $$ ] $112=3000.000 ;Z-axis maximum rate, mm/min
[21:20:43] [ $$ ] $120=200.000 ;X-axis acceleration, mm/sec^2
[21:20:43] [ $$ ] $121=200.000 ;Y-axis acceleration, mm/sec^2
[21:20:43] [ $$ ] $122=200.000 ;Z-axis acceleration, mm/sec^2
[21:20:43] [ $$ ] $130=300.000 ;X-axis maximum travel, millimeters
[21:20:43] [ $$ ] $131=300.000 ;Y-axis maximum travel, millimeters
[21:20:43] [ $$ ] $132=300.000 ;Z-axis maximum travel, millimeters
[21:20:43] [ $$ ] ok
[21:20:43] [ $I ] [VER:1.1f.20180917:]
[21:20:43] [ $I ] [OPT:VHSW]
[21:20:43] [ $I ] [MSG:Mode=STA:SSID=:Status=Not connected:IP=0.0.0.0:MAC=30-AE-A4-55-C8-28]
[21:20:43] [ $I ] [MSG:Mode=AP:SSDI=GRBL_ESP:IP=192.168.0.1:MAC=30-AE-A4-55-C8-29]
[21:20:43] [ $I ] ok
[21:20:46] [ connect ] Firmware Detected: grbl version 1.1f on COM4
```

Works for me (CONTROL v1.0.183)