OpenBuilds/OpenBuilds-CONTROL #101: Possible memory leak in viewer.js>animate()

openbuilds-engineer, 2019-09-02:
https://stackoverflow.com/questions/16072699/nodejs-settimeout-memory-leak

Requires investigation. 

apparently if you reference the timeout object in the scope of the function nodejs will do the garbage collect that correctly.
```
function redx(){
      var t = setTimeout(function(){ redx() },50);
      console.log('hi');
}

redx();
```

We do something similar in animate() to limit FPS

sharmstr, 2019-09-04:
Good luck with this one.  I've been following on the forums so before I left for the long weekend I loaded the sample hello world file.  Control has had it loaded for 5 days on my main computer (think several programs running at all times... fusion, photoshop, 100 chrome tabs, etc) and the mem usage for control has actually gone down.  Last week it was peaking at 150mb.  Now its peaking at 120mb.  Go figure.

openbuilds-engineer, 2019-09-17:
Confirmed NOT to be the real issue.  
Did fix the scoping of the animate function in 1.0.187 but that wasnt causing any issues, just not "best practice"  - either way thats been fixed, but still sure the original user's PC is the deeper cause (apart from a total rethink on the 3D viewer to really cater for low RAM PCs)

closed 2019-09-17 by openbuilds-engineer