OpenBuilds/OpenBuilds-CONTROL #102: Issue Node version

Mjay86, 2019-09-14:
Hi Team,

I try to install in on my Pi but it seems to have a version problem and i was wondering if i can get some help with that. I haven't got a lot of experience with the PI.
I followed all the instruction and i'm getting the error :

Error: The module '/home/pi/Desktop/OpenBuilds-CONTROL/node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 69. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
at Object.Module._extensions..node (internal/modules/cjs/loader.js:718:18)
at Module.load (internal/modules/cjs/loader.js:599:32)
at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
at Function.Module._load (internal/modules/cjs/loader.js:530:3)
at Module.require (internal/modules/cjs/loader.js:637:17)
at require (internal/modules/cjs/helpers.js:22:18)
at bindings (/home/pi/Desktop/OpenBuilds-CONTROL/node_modules/bindings/bindings.js:81:44)
at Object.<anonymous> (/home/pi/Desktop/OpenBuilds-CONTROL/node_modules/@serialport/bindings/lib/linux.js:1:98)
at Module._compile (internal/modules/cjs/loader.js:689:30)
at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
TypeError: Cannot read property 'address' of undefined
at Server.<anonymous> (/home/pi/Desktop/OpenBuilds-CONTROL/index.js:38:43)
at Object.onceWrapper (events.js:277:13)
at Server.emit (events.js:194:15)
at emitListeningNT (net.js:1311:10)
at process._tickCallback (internal/process/next_tick.js:63:19)
TypeError: Cannot read property 'address' of undefined
at Server.<anonymous> (/home/pi/Desktop/OpenBuilds-CONTROL/index.js:42:43)
at Object.onceWrapper (events.js:277:13)
at Server.emit (events.js:194:15)
at emitListeningNT (net.js:1311:10)
at process._tickCallback (internal/process/next_tick.js:63:19)

openbuilds-engineer, 2019-09-16:
As per the first line of the wiki:

"Instructions below are provided as-is and expects the user to be fluent in Linux, Electron, NodeJS and Git commands. No support or assistance will be provided, this instructions is for those who want to venture where there is no support. You are on your own! (: Have fun!"

As Pi is not an officially supported platform, we dont spend time supporting it

If you do figure it out, revert with a solution and we'll update the docs though - that page is community driven

openbuilds-engineer, 2019-09-16:
sidenote make sure you are using Node 8 - not 10...

closed 2019-09-16 by openbuilds-engineer

eleleung, 2020-05-15:
Hi, not sure if you solved this but someone might find this information useful since I encountered the same issue - make sure you are running Node 8 and then run `npm rebuild` after the initial install.

petervanderwalt, 2020-05-16:
Node 8 is specified in the wiki instructions

[image]

node rebuild depends on specifics (could npm install build serialport, or did it use a precompiled one?) - will add though, won't hurt to rerun it

petervanderwalt, 2020-09-02:
https://github.com/OpenBuilds/OpenBuilds-CONTROL/wiki/Installing-on-Pi4 updated instructions for Buster, Node12 etc