OpenBuilds/OpenBuilds-CONTROL #103: Display issues

Blanshan91, 2019-10-07:
I am new to this forum and playing around with this controller software for the first time and wanted share something I am seeing.

Problem-1: I used SheetCam to generate a few cut paths for a plasma table application, I drew a square and a circle.  I used the grbl plasma post processor and generated the Gcode "in inches".  I loaded the G-code into OpenBuilds Control v1.0.190 and on the bottom display it is not displaying the circle, however when I run the code the my motors follow the circle path as it should.   Is there something wrong with the software display not drawing the circle based on the G-code?
[image]
[image]

Problem-2:  When running the code (all inch mode), the software tool is not plotting in the same location as the tool path.  Is this a scaling issues with the software (inch/mm)?
[image]

openbuilds-engineer, 2019-10-08:
Post the gcode file please

Blanshan91, 2019-10-08:
G0Z0.25
G0X6.1572Y1.8153
G0Z0.15
G0Z0.15
M3
G04 P0.5
F30.0
G1X6.1572Y1.8153Z0.06
F90.0
G2X6.1572Y1.7248I-0.0453J-0.0453
G2X6.1572Y1.7248I-0.7743J0.7743
G2X6.0667Y1.7248I-0.0453J0.0453
M5
G0Z0.25
G0X3.2942Y2.4542
G0Z0.15
G0Z0.15
M3
G04 P0.5
F30.0
G1X3.2942Y2.4542Z0.06
F90.0
G2X3.3582Y2.3902I0.0J-0.064
G1X3.3582Y1.4585
G1X1.3134Y1.4585
G1X1.3134Y3.322
G1X3.3582Y3.322
G1X3.3582Y2.3902
G2X3.2942Y2.3262I-0.064J0.0
M5
G0Z0.25
G0X0.1564Y4.667
G0Z0.15
G0Z0.15
M3
G04 P0.5
F30.0
G1X0.1564Y4.667Z0.06
F90.0
G2X0.2204Y4.603I0.0J-0.064
G1X0.2204Y0.25
G3X0.25Y0.2204I0.0295J0.0
G1X7.3418Y0.2204
G3X7.3713Y0.25I0.0J0.0296
G1X7.3713Y4.603
G3X7.3418Y4.6326I-0.0295J0.0
G1X0.25Y4.6326
G2X0.186Y4.6966I0.0J0.064
M5
G0Z0.25
M5 M30

openbuilds-engineer, 2019-10-09:
Quick test to see if its the Gcode or how we read it, checked it in a different viewer (ncviewer.com) and also doesnt show up there:

So something in the gcode output is not 100% to spec.  Close enough that Grbl runs it, but its not "to standard" somewhere.   So no bug to fix here at this time:
[image]

Closing issue, but you can continue in this thread, post some more details of your post processor, etc

closed 2019-10-09 by openbuilds-engineer

openbuilds-engineer, 2019-10-09:
Problem-2: When running the code (all inch mode), the software tool is not plotting in the same location as the tool path. Is this a scaling issues with the software (inch/mm)? - added this one to the buglist... will fix in next release

comqdhb, 2020-12-28:
G21 ; mm-mode

M4 S1000; Dynamic Power Laser On

G0 X40 Y250 F1000
G2 F750 X40 Y250 I200 J0
M5
works on NCViewer, processes on openbuild arco system but does not draw a circle in the openbuild viewer.

petervanderwalt, 2020-12-28:
Please do not post in closed/old threads: This one was closed in Oct 2019 already, barely noticed the new post 

NC Viewer also does not plot your gcode:  That is probably not valid G2/3 arc code: See http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g2-g3

[image]

petervanderwalt, 2020-12-28:
[image]
 ncviewer does not plot it.  Also errors out - something is a little of out spec