OpenBuilds/OpenBuilds-CONTROL #104: OpenbuildsControl CPU Usage / Unresponsive

devsys, 2019-10-08:
Hello, I was about 5.5hrs into a 6.5hr cut and all of a sudden the machine started moving very very slow. It would dwell in one place for a while before getting another command and moving a bit. This progressively got worse and worst to the point that the machine (LEAD 1010) stopped moving alltogether. I suspected the CPU was pegged out as I've seen similar jerky behavior sending grbl to a 3d printer with a rasberry pi that had too much load on it. 

I attempted to pause and unpause with no luck. 

I see this in the log:

[image]

[image]

[image]

I ended up having to kill the job off because it was not moving.

Even after I completely exited openbuilds control and then restarted it, it was consuming a solid 7% of my CPU. 

If there are any other logs or anything else I can provide please let me know. Thanks

openbuilds-engineer, 2019-10-09:
Re your Facebook post https://www.facebook.com/groups/OpenBuilds/permalink/664887547367701/

You will be able to run large jobs, but we will limit it to say the first 500,000 lines that get rendered in the preview. Thats what eats the memory, those massive files take a lot of ram to draw the 3D preview, so just the viewer will be capped to stop excessive ram... The gcode will still run all of it (;
Ps. If you had follow up questions post here (instead of FB) thats what git is for.

openbuilds-engineer, 2019-10-09:
Closed as its already in the works... but that just takes it off the list begging for immediate attention.  Go ahead and reply if you need any more specifics

closed 2019-10-09 by openbuilds-engineer

devsys, 2019-10-09:
Thanks for the great work. I wonder if an option to disable the 3d preview for folks who are limited on cycles would be a good option in the future? Lots of folks including myself use old hardware to run the CNC due to dust/vibrations, etc...

openbuilds-engineer, 2019-10-10:
We actually already disable the 3D viewer on machines that are really too old (by checking if they support WebGL.) - of course a PC like yours is a little better than that so it does start the 3D viewer.  You'll be able to use it all fine with files with less than say 700k-1million lines of gcode... its those big jobs that cause the issues. 

Always optimising that but phew Vectric makes some crazy massive files! Still playing catchup on that. 

The best solution to cater for everyone (slow and fast PCs) is to ingest only as many gcode lines into the viewer, as dictated by the RAM allocated to the process by the OS, and leave some overhead

devsys, 2019-10-17:
You rock! Thanks! I'll grab the update and give that a shot.

Moonlight63, 2020-09-09:
Sorry to resurrect. This seems to be like where I should post. Using Control on a Linux Mint machine, if I full screen the app, it crashes and I have to force close it. it worked fine for a while initially, then randomly started happening consistently. Only seems to happen on my laptop. Running newest v246 on Mint 19. I'll see if I can record a video of the behavior. I disabled all the 3d stuff but it still happens.