OpenBuilds/OpenBuilds-CONTROL #106: Check USB Selective Suspend setting programatically

openbuilds-engineer, 2019-11-13:
https://www.npmjs.com/package/windows-registry

petervanderwalt, 2020-04-20:
powercfg /GETACTIVESCHEME
Power Scheme GUID: 381b4222-f694-41f0-9685-ff5bb260df2e  (Balanced)

C:\Users\user>Powercfg.exe -setacvalueindex 381b4222-f694-41f0-9685-ff5bb260df2e 2a737441-1930-4402-8d77-b2bebba308a3 48e6b7a6-50f5-4782-a5d4-53bb8f07e226 0

petervanderwalt, 2020-04-20:
powercfg /q <active guid>

...
  Subgroup GUID: 2a737441-1930-4402-8d77-b2bebba308a3  (USB settings)
    Power Setting GUID: 48e6b7a6-50f5-4782-a5d4-53bb8f07e226  (USB selective suspend setting)
      Possible Setting Index: 000
      Possible Setting Friendly Name: Disabled
      Possible Setting Index: 001
      Possible Setting Friendly Name: Enabled
    Current AC Power Setting Index: 0x00000000
    Current DC Power Setting Index: 0x00000001
...

petervanderwalt, 2020-04-20:
will be in 1.0.234 or later
[image]

closed 2020-04-20 by petervanderwalt