OpenBuilds/OpenBuilds-CONTROL #108: goto MCS need susable settable offsets

swarfer, 2019-12-09:
When homing and hard limits are enabled sending the command
G53 G0 Z0  (and X and Y)
will cause the limit switch to be hit.

The default setup needs to have suitable defaults for a negative offset:
G53 G0 Z-5
and the user ,must be able to set the offsets for X,Y,Z to suite their machine.

'-5' above is an example.  a good default value will need to be determined for the OB limit switch kits.

petervanderwalt, 2020-06-22:
Suggestion, shall we just read Grbl settings> Homing Pull off and then park at that spot? Z-<homingpulloff>

swarfer, 2020-06-22:
that is a good idea if homing is enabled.
what is the GRBL default if homing not enabled?  may still be a safe offset
from machine home.

petervanderwalt, 2020-06-22:
If yiu have limits, why not have homing. G53 Z0 would not crash into non
existant limits for machines without limits

swarfer, 2020-06-23:
On Mon, Jun 22, 2020 at 9:46 PM Peter van der Walt <notifications@github.com>
wrote:

> If yiu have limits, why not have homing. G53 Z0 would not crash into non
> existant limits for machines without limits
>

generally true, I think this will work

petervanderwalt, 2020-06-23:
Actually, working on the code today reminded me: we already disable Go to Machine zero entries in the menu if $22=0 (: So no worries about people who don't home! (: 

https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/54edbcb6fd2b5aab48b58e51c181d4dc6b8b5b75/app/js/grblsettings.js#L67-L79

petervanderwalt, 2020-06-23:
Note to self, also implement same button/js into mobile jog pendant

swarfer, 2020-06-23:
On Tue, Jun 23, 2020 at 2:26 PM Peter van der Walt
<notifications@github.com> wrote:
>
> Actually, working on the code today reminded me: we already disable Go to Machine zero entries in the menu if $22=0 (: So no worries about people who don't home! (:
>
> https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/54edbcb6fd2b5aab48b58e51c181d4dc6b8b5b75/app/js/grblsettings.js#L67-L79

and that is exactly why I don't use CONTROL as my daily driver, I had
to create macros to 'go home before switch off' on my switchless
machine.
going to Z5 before XY moves pretty much always hits some of the crap
lying on the table so I cant use that.

-- 
-- 
David the swarfer      http://www.swarfer.co.za

petervanderwalt, 2020-06-23:
> going to Z5 before XY moves pretty much always hits some of the crap

But months ago you had me add this one for you:  (G53 Z0 first, then XY0)  to avoid that already

[image]

petervanderwalt, 2020-06-23:
But aahhh, I see... Because you don't have limits, only "fake" limits, thats not an option.  
(; why can't you be normal lol!    Fit switches or not lie about switches pick one lol!

swarfer, 2020-06-23:
On Tue, Jun 23, 2020 at 7:45 PM Peter van der Walt <notifications@github.com>
wrote:

> going to Z5 before XY moves pretty much always hits some of the crap
>
> But months ago you had me add this one for you: (G53 Z0 first, then XY0)
> to avoid that already
>

disabled when no home switches <-:
but faking the home makes it a perfectly valid and safe move

-- 
David the swarfer      http://www.swarfer.co.za

swarfer, 2020-06-23:
> But aahhh, I see... Because you don't have limits, only "fake" limits,
> thats not an option.
> (; why can't you be normal lol! Fit switches or not lie about switches
> pick one lol!
because faking the home the way I do and advise anyone without switches to
do, is a working and safe solution.
the only thing better would be if soft limits also worked without homing
enabled, because that would encompass the entire needs of the machine:
a) a defined home
b) soft limits to prevent crashing into the other end of travel.

hmmm, maybe I should hack GRBL a bit (-:

petervanderwalt, 2020-06-23:
> hmmm, maybe I should hack GRBL a bit (-:

The predicament is we kinda have to accomodate 
- all stock grbl
- no config needed except Grbl settings (we react based on what we can read from those settings) 
- 90% of the machine bundles ship out with switches, unlike the old days where no one installed switches on hobby machines
- Faking the home, you already know how much I dislike it.  It really boils down for us to a) you have switches, great, home, and then you can use G53 moves. If not, Setzero and stick to G54.  Know what you are doing, use G55-59.1 etc but we can't do much handholding, CONTROL is for beginners

petervanderwalt, 2020-06-23:
Open to suggestion to handle G53 Z0 moves without asking the user if he's "faked" the home instead of counting on Grbl settings. But I can't come up with any (and don't want to enable G53 for everyone, you know exactly what the newbies will do)

petervanderwalt, 2020-06-23:
[image]

Update for those with Limits enabled - with offsets based on Homing Pull-Off (Beta)  (set to 5.000mm in my test profile here, you can see the G53 G0 XYZ-5.000)

closed 2020-06-23 by petervanderwalt