OpenBuilds/OpenBuilds-CONTROL #113: Bug: G20/G21 can cause render & simulate to use different units than machine moves.

batcrave, 2020-03-03:
(originally mentioned in [https://openbuilds.com/threads/openbuilds-control-software.13121/page-37#post-93589](url) )

 -    Loading a program with unspecified units (no G20/G21 statements) will render in the 3D view (and, by default, cut) as if it were in millimeters, regardless of which unit (inch/mm) mode the UI is operating in. This isn't a major problem in itself, but...

-    Switching from metric to inches with a G20 in the console (with or without hitting "Update 3D View" afterwards), will then cause the machine to _move_ in inches, while still displaying the mm-scale render and simulating at the rendered scale.

-    Oddly enough, _ending_ the program with a G20 will cause it to render in inches, while _beginning_ the program with a G20 and ending with a return to G21 will render and simulate at mm scale (the opposite of expected behavior) while cutting based on whichever G20/G21 mode is set in the console. The renderer appears to use the last-detected G20/21 as the render & sim scale for the entire program, rather than considering where it occurs.

-   Getting into slightly silly territory, alternating between G20 & G21 statements in a program causes all sorts of confusion when it comes time to run, with drastic speed changes (F100 means very different things depending on whether you interpret it as in/m or mm/m) in addition to the obvious differences in distance. Hopefully the number of edge cases who actually try to switch units back and forth within a single program will be extraordinarily few, though - most of the scenarios I can think of involve hand-written code & canned cycles, which probably isn't CONTROL's key demographic - so this is probably the least hazardous manifestation of the problem.

petervanderwalt, 2020-03-19:
https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/master/app/lib/3dview/workers/litegcodeviewer.js#L941-L950 relates, parser kinda hopes whole file is in one format 
Someday a lot of work can be done here to check when nit changes (i think there is leftover code here https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/master/app/lib/3dview/workers/litegcodeviewer.js#L237-L245 ) and scale inch-mode sections in here already (right now we scale afterwards)

petervanderwalt, 2022-04-21:
Closing issue - unlikely to be fixed and the use case for G20 and G21 in a single job file is extremely rare.

closed 2022-04-21 by petervanderwalt