OpenBuilds/OpenBuilds-CONTROL #116: CONTROL software loaded GCode file name

rmarshall71, 2020-03-23:
I would like to request that when a GCode file is loaded in CONTROL software that the name of the file be displayed somewhere on the screen. When I do a 2 bit carve, some of the output displays look very similar.

rscamp, 2020-03-23:
> I would like to request that when a GCode file is loaded in CONTROL software that the name of the file be displayed somewhere on the screen. When I do a 2 bit carve, some of the output displays look very similar.

Thanks for requesting this, Ron.

petervanderwalt, 2020-03-30:
Implemented in 1.0.220 or later

petervanderwalt, 2020-03-30:
[image]

closed 2020-03-30 by petervanderwalt

rmarshall71, 2020-03-31:
Thank you. When will the update be released? I am running 1.0.215 and it doesn't find the update yet. No hurry, just asking.

petervanderwalt, 2020-03-31:
In a day or two. Internal testing, management approval etc (:

rmarshall71, 2020-03-31:
Like I said, no hurry. I appreciate you adding this feature. I love the software.

Ron