OpenBuilds/OpenBuilds-CONTROL #118: Bug: Autoupdate still autoupdates when told not to (auto?)update

batcrave, 2020-03-24:
Quitting the app after clicking "Not right now" on the update dialog will still perform an update as soon as the user quits. I've confirmed this behavior on both the Windows version (on Win 7) and the linux AppImage version (on Debian stretch) - I assume it's also true for the .deb, but haven't tested it. [Peter confirms ](https://openbuilds.com/threads/openbuilds-control-software.13121/page-38#post-94306) that this is not the intended behavior (right before telling me I'm being deprioritized because I live in the filthy linux ghetto and use a terrible window manager that everyone who knows better hates *sniffle*) (not that I necessarily disagree).

This is the stdout from 1.0.207 launching, opening the window, letting the timer expire & download, clicking "Not right now" in the dialog", and then, shortly after, closing the window & quitting:

```
bats@puzzlebox:~/OpenBuildsCONTROL$ ./OpenBuildsCONTROL-1.0.207.AppImage
Starting OpenBuilds CONTROL v1.0.207
Local User Data: /home/bats/.config/OpenBuildsCONTROL
Command Line Arguments for Electron: Set OK
https: listening on:10.0.0.212:3001
http: listening on:10.0.0.212:3000
Fontconfig warning: "/etc/fonts/fonts.conf", line 100: unknown element "blank"
Created Uploads Temp Directory
Checking for update
Found version 1.0.215 (url: OpenBuildsCONTROL-1.0.215.AppImage)
{"version":"1.0.215","files":[{"url":"OpenBuildsCONTROL-1.0.215.AppImage","sha512":"RSkJcHjwCa5IoERFjYns/RrsnM4vEgs52Hg/Mriwf5M1rBrFEp+HVftfND/7xBmbjRpk8D/CsQflwlM8ybGo+w==","size":100396018,"blockMapSize":105478}],"path":"OpenBuildsCONTROL-1.0.215.AppImage","sha512":"RSkJcHjwCa5IoERFjYns/RrsnM4vEgs52Hg/Mriwf5M1rBrFEp+HVftfND/7xBmbjRpk8D/CsQflwlM8ybGo+w==","releaseDate":"2020-03-20T14:13:08.565Z","releaseName":"v1.0.215","releaseNotes":""}
Downloading update from OpenBuildsCONTROL-1.0.215.AppImage
updater cache dir: /home/bats/.cache/openbuildscontrol-updater
Cached update file doesn't exist, directory for cached update will be cleaned
jcoaC2XHS01kxOeYX3atjgko duplicated in blockmap (same size), it doesn't lead to broken differential downloader, just corresponding block will be skipped)
oypxBdLHAOvlbVxZm4C9vbkA duplicated in blockmap (same size), it doesn't lead to broken differential downloader, just corresponding block will be skipped)
File has 4751 changed blocks

[ ... kind/start/end blocks snipped ... ]

Full: 98,042.99 KB, To download: 96,990.16 KB (99%)
Differential download: https://github.com/OpenBuilds/OpenB...d/v1.0.215/OpenBuildsCONTROL-1.0.215.AppImage
download range: bytes=159006-70653604
Redirect to https://github-production-release-a...37859794/6becbd00-6ac5-11ea-8d27-4317ba000d2e
download range: bytes=70759516-89048372
download range: bytes=89143118-89314962
download range: bytes=89389741-89960675
download range: bytes=90049096-90230239
download range: bytes=90321861-90623782
download range: bytes=90708696-95428196
download range: bytes=95460159-99849138
download range: bytes=100090393-100290535
New version 1.0.215 has been downloaded to /home/bats/.cache/openbuildscontrol-updater/pending/OpenBuildsCONTROL-1.0.215.AppImage
exit
Auto install update on quit
Install: isSilent: true, isForceRunAfter: false

bats@puzzlebox:~/OpenBuildsCONTROL$
```

petervanderwalt, 2020-03-25:
```
Peter confirms that this is not the intended behavior (right before telling me I'm being deprioritized because I live in the filthy linux ghetto and use a terrible window manager that everyone who knows better hates sniffle) (not that I necessarily disagree).
```

:D shame! 

Well I have bad news for you,  i feel like thats wrong too it shouldnt update on a normal quit, only on a "do it" click.  But sadly, see the last sentence of https://www.electronjs.org/docs/api/auto-updater#event-before-quit-for-update:

```
autoUpdater.quitAndInstall()
Restarts the app and installs the update after it has been downloaded. It should only be called after update-downloaded has been emitted.

Under the hood calling autoUpdater.quitAndInstall() will close all application windows first, and automatically call app.quit() after all windows have been closed.

Note: It is not strictly necessary to call this function to apply an update, as a successfully downloaded update will always be applied the next time the application starts.
```

"a successfully downloaded update will always be applied the next time the application starts." sounds like the part irking us?

batcrave, 2020-03-26:
> :D shame!
Fine, fine, you're right... you actually told me some of that _before_ confirming the intention :)
(and I really can't disagree on most of your reasoning, either)

> _"as a successfully downloaded update will always be applied the next time the application starts."_
Yep, I think you just nailed out problem. That's... really really pretty awful. Not sure how they thought that sounded like a good idea, unless they were expecting it to be called in a very different way.

With that in mind, I'm still trying to parse exactly how the code handles the update process (I'm beyond rusty at this), but is there any reason not to split it up so checkUpdate() just sends an "I'm done", which will trigger the dialog, and then, only  after getting an affirmative click, trigger downloadUpdate?

ie (assuming my awkward pseudocode is any more readable than my prose), instead of:

`checkUpdate() -> updateTime(){downloadUpdate} ->autoUpdater.checkForUpdates()-> [dialog:  yes?{applyUpdate} || no?{tooLateApplyUpdateAnyhowSucker}]`

do something like :
` checkUpdate() -> updateTime(){"ok, I'm done"} -> [dialog: yes?{downloadUpdate, etc} || no?{noProblem,doNothing}]`

That way it still checks for the update & pesters the user, but the whole unstoppable chain of events that `update-downloaded` apparently sets rolling is only unleashed once it gets a positive click from the dialog, rather than being set loose immediately on discovering the new version.

Or am I overlooking some reason this wouldn't work?

petervanderwalt, 2020-03-26:
Sadly, again, they are forcing your hand:

```
Event: 'update-available'
Emitted when there is an available update. The update is downloaded automatically
```

(: but I have a plan:
- use github api to see if there's a newer version, then use autoUpdater.checkForUpdates()

batcrave, 2020-03-26:
> Sadly, again, they are forcing your hand:

I'm starting to really hate them for this. I think they may even be doing it deliberately.

But I guess I hadn't looked closely enough - I'd thought most of update.js's checkUpdate() stuff was independent of the autoupdater until it emitted the downloadUpdate and could been used safely, without them poking their damn autoinstalling fingers in & making a mess of it. 

Sort of like if you were to...
> use github api to see if there's a newer version, then use autoUpdater.checkForUpdates()

Brilliant!
Hah! Take _that_, autoUpdater!

(although maybe I better not antagonize it too much)

petervanderwalt, 2020-03-26:
No you may (:
More options is better, default to that if you want to but at least add an
option to the API. Other than that though it's brilliant. Give it a Github
Releases url, and use Github releases (which we love too as we can automate
releases and compiles with TravisCI and AppVeyor) and from there it does
the rest. No need for seperate systems for seperate OSs, no need to host
your own Squirrel server like it's predecessor, etc. Pros and cons. At
least I see a workaround for the single niggles we have with it

Ps: I think you are right, checkupdate already uses git api, haven't worked
on that in ages (probably 3 years or so, updates just 'worked' so far)

petervanderwalt, 2020-03-30:
Altight tried the other way round, but its painful for everyone who doesnt sit on gigabit fiber. 
I click "proceed with update" then spend an hour waiting for the download 

Sticking to the existing behaviour - will add a note that it will also update when you quit though

closed 2020-03-30 by petervanderwalt