OpenBuilds/OpenBuilds-CONTROL #120: Running in browser?

JohnInWI, 2020-04-10:
Forgive my noobness.

I am trying to run the CONTROL software right in Chrome on a Windows 10 computer. When running in Chrome, I get "waiting for USB" despite there being USB comm ports available.
The Electron exe CONTROL app works great, but I was hoping to do a couple tweaks.

It's not completely clear to me how & where the websocket server is set up. I do see a var for an IP address in websocket.js that I'm not sure if I need to change. Am I barking up the right tree?

petervanderwalt, 2020-04-10:
Check the Ctrl+Shift+I log > Console tab in Chrome for clues
Likely some plugin blocking JS, or SocketIO

petervanderwalt, 2020-04-10:
```I do see a var for an IP address in websocket.js  ```
Commented out on line 2? Not in use, commented out.

JohnInWI, 2020-04-10:
Thanks for the responses.
Yes, when the backend is running, I can go to  http://ip-address-of-pc-running-chrome:3000/ and successfully run CONTROL in Chrome.

I was trying to just run index.html rather than the entire Electron app. I will set it up as an Electron app and mod from there (I'm just changing the grbl baud rate)

Again, noob.

closed 2020-04-13 by petervanderwalt