OpenBuilds/OpenBuilds-CONTROL #128: Misc.: No +/- in inch mode ; Probe success dialog focus ; Desktop/mobile dialog sync ;  "Rendering, please wait"

batcrave, 2020-05-12:
(all current circa .234)

1. The increase/decrease jog increment key bindings only work in mm mode, not in inches (although the rest of the bindings still seem to operate normally). This may be a long-standing issue, since I have notes from a month or so back about +/- being intermittently broken (although at the time I didn't manage to associate it with the mode)

2. After using the Clear button in the g-code editor (or manually deleting all the code & then hitting update), the 3D View tab's title sits with an endless progress spinner on "rendering, please wait" until some code (movement codes, specifically) is entered & the view is updated again. This is probably just cosmetic.

3. The OK button on the probe success dialog doesn't have focus (unlike most of the other dialogs now - thanks for that, btw)

4. Error dialogs (and the clearing thereof) isn't synced between desktop & mobile interface. So it's possible to clear a dialog on one and have it still sitting on the other - then, when you go to clear the second one, it resets everything all over again. Particularly a problem if you start a program running on desktop, then pick up the tablet & walk away, only to later discover that it's occupied by a (now obsolete and irrelevant) dialog you can't get rid of without wrecking the whole process.

petervanderwalt, 2020-05-16:
> 4\. only to later discover that it's occupied by a (now obsolete and irrelevant) dialog you can't get rid of without wrecking the whole process

In the meantime, a quick F5/refresh clears open dialogs (;

batcrave, 2020-05-16:
> In the meantime, a quick F5/refresh clears open dialogs (; - mobile and app both

But... but... but... that's... too _easy!_

(seriously, though, I thought sure I'd tried that on the mobile version - and feel more than a little stupid if I didn't - but I had absolutely no idea it would work on the desktop. that'll come in handy)

petervanderwalt, 2020-06-22:
> The increase/decrease jog increment key bindings only work in mm mode, not in inches (although the rest of the bindings still seem to operate normally). This may be a long-standing issue, since I [...]

Fixed in 1.0.238

petervanderwalt, 2020-06-22:
> After using the Clear button in the g-code editor (or manually deleting all the code & then hitting update), the 3D View tab's title sits with an endless progress spinner on "rendering, please wait" [...]

Fixed in 1.0.238 or later

petervanderwalt, 2020-06-22:
>The OK button on the probe success dialog doesn't have focus (unlike most of the other dialogs now - thanks for that, btw)

Fixed 1.0.238

petervanderwalt, 2020-06-22:
> Error dialogs (and the clearing thereof) isn't synced between desktop & mobile interface. So it's possible to clear a dialog on one and have it still sitting on the other - then, when you go to [...]

Moving to new issue - no sync exists yet

closed 2020-06-22 by petervanderwalt

petervanderwalt, 2020-06-22:
https://github.com/OpenBuilds/OpenBuilds-CONTROL/issues/136 refers