OpenBuilds/OpenBuilds-CONTROL #129: Macro shortcuts don't check for conflicts

batcrave, 2020-05-16:
As the title implies, it's possible to assign ctrl-alt-shift-meta-opt-cmd-* to more than one macro, leaving it ambiguous whether the user will get the action they're expecting when they hit it. It seems to use whichever was assigned first, but that's probably little consolation when the user wanted to run the _other_ one they'd accidentally assigned it to.

( *ok, fine, so technically it _won't_ allow you to use meta as a modifier key. DAMHIK)

batcrave, 2020-05-16:
Related issue: There doesn't seem to be any way to remove or clear a shortcut from a macro.

petervanderwalt, 2020-06-22:
> doesn't seem to be any way to remove or clear a shortcut from a macro. 

Activated Metro's clear button in 1.0.238 or later

> As the title implies, it's possible to assign ctrl-alt-shift-meta-opt-cmd-* to more than one macro,  

Added 1.0.238

closed 2020-06-22 by petervanderwalt