OpenBuilds/OpenBuilds-CONTROL #131: DRO moves happen at rapid rate, rather than jog rate

batcrave, 2020-05-16:
I'm not sure if this really counts as an _issue_, or if my brain just doesn't work like CONTROL works, but every time I go to move an axis by typing in the DRO, I'm inevitably surprised when it moves so much faster than the rates I've been jogging around at. 

There may be some perfectly grbly reason for not handling them _as_ jog moves, but it seems - at least to my batty little brain - like the more intuitive setup would be, instead of G0, to use "G1 F[jog rate]", seeing as the setting is sitting right next to the DROs and is the rate that any other manual movement (console users excepted) is going to happen at.

batcrave, 2020-05-16:
(of course, the _really_ intuitive setup, to my mind, would be for changes to the DRO to edit the current WCS rather than _moving_ anywhere... but that may be my Mach 3 talking)

batcrave, 2020-05-16:
Also, a related (minor) UI note:
Editing an X/Y/Z DRO is only possible after clicking on the numbers themselves, rather than anywhere in the DRO, as seems more intuitive. A wider text box - extending all the way to the edge of (or including) the X/Y/Z label would be preferable.

petervanderwalt, 2020-06-22:
> Editing an X/Y/Z DRO is only possible after clicking on the numbers themselves, rather than anywhere in the DRO, as seems more intuitive. 
Fixed in 1.0.238 or later

> There may be some perfectly grbly reason for not handling them _as_ jog moves, but it seems - at least to my batty little brain - like the more intuitive setup would be, instead of G0, to use "G1 [...]

I get your logic, but I think the logic was the DRO moves is a "move to" not a "move incrementally toward" - therefore G0 instead of $J (never G1 for jog! - Grbl has https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging ) 

I can use G90 jogs though, so will keep ticket open for now and do that someday soon

closed 2020-06-22 by petervanderwalt