OpenBuilds/OpenBuilds-CONTROL #132: Custom probe plate thickness setting reverts to 20mm on restart

batcrave, 2020-05-19:
After quitting and restarting the software, the Probe->Advanced->Plate Thickness setting reverts from whatever custom setting had previously been in use to 20mm. 

This can be anywhere from inconvenient to minorly disastrous, depending on how much attention the user is paying.

 I _think_ I've also caught it reverting once or twice without a restart, but I can't confirm that, never mind repro it.

petervanderwalt, 2020-06-22:
did a quick test over the weekend, and it saves for me.

Couple questions:

- Does X and Y also revert for you?
- Ctrl+Shift+I > Devtools > Console Tab > Enter ```localStorage.getItem('customProbe')``` , press enter, Screenshot

petervanderwalt, 2020-06-22:
Nevermind, fixed.  Z Probe... Not Custom XYZ   - didnt notice that (;  screenshot next time man (; 

V1.0.238 and up

petervanderwalt, 2020-09-15:
https://openbuilds.com/threads/openbuilds-control-software-with-z-probe.16385/ says its back

petervanderwalt, 2020-10-12:
v1.0.251

closed 2020-10-12 by petervanderwalt