OpenBuilds/OpenBuilds-CONTROL #133: Continuous jog doesn't discontinue when textbox gains focus

batcrave, 2020-06-04:
Moving focus (by clicking or tabbing) into a textbox (jog rate or console), while moving in continuous jog mode will cause the current movement to continue continuously - presumably up to the full 1000mm - unless/until focus is moved out of the textbox and another (continuous) jog command is sent. This almost crashed me a couple times the other day, before I figured out what was going on.

(this is current as of .237, but I don't know how far back it goes)

petervanderwalt, 2020-06-22:
So, how do you do this dance?

Hold continuous with one hand on keyboard, then use mouse in other hand to click on Feedrate? 

If you were clicking jog button, there'd be a mouseup event before you can mousedown in the Feedrate input?  Had a jog key down - how did you get to feedrate without letting the button go?  So you must be doing something too advanced (;

petervanderwalt, 2020-06-23:
Closing for now, will re-open when you reply (:

closed 2020-06-23 by petervanderwalt