OpenBuilds/OpenBuilds-CONTROL #141: Independent Z Jog Setting

Sindarius, 2020-07-10:
It would be nice to have an independent z jog setting separate from X/Y. More than once I have accidentally hit page down when jogging and driven my bit into my work piece or wasteboard. 90% of the time when I am moving Z I am only looking to jog it 0.1 or 1mm at a time to carefully move it up and down but want to be able to move X/Y 10mm+ or continuously. It's nice to be able to protect me from myself sometimes.

brrian11, 2020-07-11:
I second this.

petervanderwalt, 2020-07-14:
Not at this time, tune your Z acceleration/max rate appropriately and apply focus while operating your machine. Good luck

closed 2020-07-14 by petervanderwalt

Sindarius, 2020-07-15:
[image]

Ended up implementing it myself.

petervanderwalt, 2020-07-15:
Good, too cluttered for new-users which is the focus of CONTROL.

brrian11, 2020-07-15:
Not that I think I'll change your mind, but FWIW I don't think that adds clutter at all & it's quite useful and safer, especially for new users.