OpenBuilds/OpenBuilds-CONTROL #159: Disable Home when unlock alarm is active

sharmstr, 2020-11-16:
Not a priority, but you should disable homing when unlock alarm is active.

petervanderwalt, 2020-11-16:
I think, Homing is allowed in Alarm, because Grbl has 

```
[MSG:'$H'|'$X' to unlock]- Alarm state is active at initialization. This message serves as a reminder note on how to cancel the alarm state. All g-code commands and some ‘$’ are blocked until the alarm state is cancelled via homing $H or unlocking $X. Only appears immediately after the Grbl welcome message when initialized with an alarm. Startup lines are not executed at initialization if this message is present and the alarm is active.
```

If Grbl boots up alarmed (homing = enabled for example) Homing is a valid recovery option from the alarm state 

To be tested of course (: 

I assume you raised the issue because something unintended happened?

sharmstr, 2020-11-16:
I'll test it again.  I happened to hit homing first, then while homing, I cleared the alarm which stopped homing.  Expected I suppose.  I'll home and not clear alarm and see if it does it on its own.

petervanderwalt, 2020-11-16:
> then while homing, I cleared the alarm which stopped homing

Yeah, fair enough the Clear Alarm does issue a Reset as well, so in a way it does an Abort too to clear all the lingering errors, etc 

See https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/c841d895ce3e2db3b2ff565568813ffc5544ff6b/index.js#L1568-L1569

I guess we could instead consider disabling Clear Alarm, if an Alarm is not present instead (:

sharmstr, 2020-11-16:
Okay.  I was going from memory which is getting worse and worse.  Anyhow, I must have not noticed that the unlock stopped flashing when it started to home.  I'm sorry for wasting your time with this.  Ugh.

petervanderwalt, 2020-11-16:
(: Never a problem!  Its good to talk things like these through, it only makes CONTROL better (:

closed 2020-11-29 by petervanderwalt