OpenBuilds/OpenBuilds-CONTROL #163: Not Executing

rockman44181000, 2020-12-22:
I've installed using your method 3 times now and while I have a basic knowledge of Linux to be dangerous. I can not get the Openbuilds Controller to Open, Execute or otherwise. I set the Video to MSK and after it finishes installing I get a message stating that everthing has been installed. Does this work with the 64Bit Version of Rapian OS or is it specifically for the 32bit Version?

dmax66, 2020-12-24:
Same problem here.
I started it from the command line to see if there were any errors, and I found one - the application complains about /opt/OpenBuildsCONTROL/resources/app.asar.unpacked/firmware.bin (EACCESS error). The file does exist, so I changed the read/write permissions to w for all (dangerous?)
After that, I tried to start it again. No error this time, but no GUI appears. The program just does nothing.

My system is Ubuntu 20.10, 64 bit

dmax66, 2020-12-24:
I've run a stack trace to figure out what is happening.

A snippet:
epoll_wait(29, [{EPOLLIN, {u32=34, u64=34}}], 1024, 0) = 1
read(34, "\2\0\0\0\0\0\0\0", 1024)      = 8
futex(0x344a4c689608, FUTEX_WAKE_PRIVATE, 1) = 1
recvmsg(55, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(55, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(56, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=11, events=POLLIN}, {fd=12, events=POLLIN}, {fd=55, events=POLLIN}, {fd=56, events=POLLIN}], 4, 0) = 0 (Timeout)
recvmsg(55, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(55, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(55, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(55, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(56, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=11, events=POLLIN}, {fd=12, events=POLLIN}, {fd=55, events=POLLIN}, {fd=56, events=POLLIN}], 4, 26170) = 1 ([{fd=12, revents=POLLIN}])
recvmsg(55, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(55, {msg_namelen=0}, 0)         = -1 EAGAIN (Resource temporarily unavailable)
read(12, "!", 2)                        = 1
epoll_wait(29, [{EPOLLIN, {u32=85, u64=85}}], 1024, 0) = 1
read(85, "{\n   \"blockdevices\": [\n      {\"n"..., 65536) = 32768
madvise(0x344a4ebd5000, 65536, MADV_DONTNEED) = 0
write(34, "\1\0\0\0\0\0\0\0", 8)        = 8
futex(0x344a4c689608, FUTEX_WAKE_PRIVATE, 1) = 1
epoll_wait(29, [{EPOLLIN|EPOLLHUP, {u32=85, u64=85}}, {EPOLLIN|EPOLLHUP, {u32=87, u64=87}}, {EPOLLIN, {u32=34, u64=34}}], 1024, 0) = 3
read(85, "og-sec\":512, \"rota\":false, \"sche"..., 65536) = 7564
madvise(0x344a4ebd5000, 65536, MADV_DONTNEED) = 0

Not able to proceed further - I could download the source code and check what is happening, but I don't want to waste my time; I'd like just to start using my CNC machine

rockman44181000, 2020-12-24:
Instead of Ubuntu, I have talked with numerous people who swear by Twister OS overlayed on Raspian OS. Apparently you can even get it to run like Windows. I had the same results with Ubuntu as well. Available here, Twister OS

| 
| 
|  | 
Twister OS

 |

 |

 |

petervanderwalt, 2020-12-28:
We do not provide technical support for the Raspberry Pi installs

closed 2020-12-28 by petervanderwalt