OpenBuilds/OpenBuilds-CONTROL #165: Dont zero out current position at beginning of probe routine

sharmstr, 2020-12-24:
Is it possible to not zero out current position at the start of probing routines?

I've been running jobs with multiple tool changes.  The probe button defaults to xyz probe.  If I forget to change to the Z tab when probing for a new tool length, it wipes out my X,Y home position even though I stopped the xyz probe routine.  Its annoying and totally understand its my fault.  Just hoping there's a better way for you to do it.

petervanderwalt, 2020-12-28:
Ahh man, sorry!

Sadly, G38.2:

"Program G38.n axes to perform a straight probe operation. The axis words are optional, except that at least one of them must be used. The axis words together define the destination point that the probe will move towards, starting from the current location."

Last few words in there "starting from the current location" - means its easier to set zero before moving (knowing that we'll anyway zero after) 

So how can we prevent you accidentally running XYZ instead (; - once its started theres kinda no going back (abort also looses position)

petervanderwalt, 2020-12-28:
What about using the context probe shortcuts as a habit? 
[image]

or a Javascript macro that runs ```openProbeZDialog()```

sharmstr, 2020-12-28:
Ah, ok.  I was hoping that there was a way to store current position into a variable like we do with the probing functions in PathPilot (linuxcnc).

I was using the context probe shortcut for awhile, but found it easier to hit the probe button instead.  The javascript macro is a good option.  I didnt realize we could call functions for the macros.  Awesome.

petervanderwalt, 2020-12-28:
JS Macros are awesome, can even do DOM manipulation: 

> Put this in a JS Macro, run it once everytime the application start to inject the button onto the toolbar

```
var probezbtntpl = `<button id="grblProbeZMenu" class="ribbon-button" onclick="openProbeZDialog();">
                  <span class="icon">
                    <span class="fa-layers fa-fw">
                      <i class="fas fa-podcast fa-rotate-180 fg-blue"></i>
                    </span>
                  </span>
                  <span class="caption grblmode">Probe Z</span>
                </button>`
$( "#grblProbeMenu" ).before( probezbtntpl  );

```
[image]

sharmstr, 2020-12-28:
Whoa.  I take a year off and missed all the good things!  Awesome.  Thank you!

closed 2020-12-28 by petervanderwalt

petervanderwalt, 2020-12-28:
> Whoa. I take a year off and missed all the good things!

https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/master/CHANGELOG.txt (; 

More on JS macros: https://openbuilds.com/threads/help-getting-started-with-javascript-macros-in-control.15880/