OpenBuilds/OpenBuilds-CONTROL #171: Close 'Completed' Dialog on Run by keyboard shortcut (spacebar)

brrian11, 2021-01-01:
The new 'Completed' dialog w/ time is nice, but since many of my jobs repeat, I'm not a fan of having an extra click to get rid of the dialog. I'd prefer the total time to be displayed in another way - maybe in the bottom left corner where the countdown timer is.

A way around the extra click, and something I use anyway, is the spacebar keyboard shortcut to rerun the job. It works and runs the job, but the 'Completed' dialog stays & CONTROL remains grayed out. Would it be possible to automatically close the dialog if a keyboard shortcut is used?

Thank you.

[image]

sharmstr, 2021-01-01:
Does enter clear it?

brrian11, 2021-01-02:
Just tested & yes, it clears it, whether it's before a keyboard shortcut is used or after.

That doesn't really change the request though - I'd still like it to dismiss itself, at least when a keyboard shortcut is run, but better, upon any user action in CONTROL.

petervanderwalt, 2021-01-11:
The real issue here, is keyboard shortcuts shouldnt run while the dialog is open... Will fix that instead, rest of behavior is as intended, for a reason

brrian11, 2021-01-11:
So I'll have to close the dialog before I can run a program, no matter how I do it? If so, may I ask what the reason is?

petervanderwalt, 2021-02-01:
Enter now closes the dialog, shouldn't be too much effort,thanks! (:

closed 2021-02-01 by petervanderwalt