OpenBuilds/OpenBuilds-CONTROL #172: Run Macro On Startup

sharmstr, 2021-01-04:
Add to the wish list the ability to set a macro to run on startup.  Assigning a keyboard shortcut to my macro makes it pretty easy to set my environment, but I have to remember to do it :)

petervanderwalt, 2021-02-01:
FYI I gave in to popular demand and changed the Increments (was one of the features I know we wanted auto run for)

1.0.277 and up:  
[image]

petervanderwalt, 2021-02-01:
v1.0.277

[image]

sharmstr, 2021-02-01:
Yes!

petervanderwalt, 2021-02-01:
prevents clicking button, and shows type on the button itself 
[image]

sharmstr, 2021-02-01:
Nice touch.  I'm using hidden elements to know if a macro's been run or not.  This makes my life easier.

sharmstr, 2021-02-18:
Hey Peter

I found an issue with this.  When you create the macro and click Apply, it runs the code.  Not sure if this should be the default.  I suppose since you cant run it by clicking the button that this behavior might make sense.  

However, adding other macros or editing existing macros, even if they are not set to run at startup, will trigger the startup macro(s) when you click apply.    Here's a log of the testing I've done.

Two macros run at startup as expected:
[02:54:43] [ macros ] Macro: SM 1 executed on startup!
[02:54:43] Startup Macro 1
[02:54:43] [ macros ] Macro: SM 2 executed on startup!
[02:54:43] Startup Macro 2

**Then add new macro that doesnt run at startup:**
[02:56:48] [ macros ] Macro: SM 1 executed on startup!
[02:56:48] Startup Macro 1
[02:56:48] [ macros ] Macro: SM 2 executed on startup!
[02:56:48] Startup Macro 2

**Then edit Startup Macro 2**
[02:57:53] [ macros ] Macro: SM 1 executed on startup!
[02:57:53] Startup Macro 1
[02:57:53] [ macros ] Macro: SM 2 executed on startup!
[02:57:53] Startup Macro 2 Edit

sharmstr, 2021-03-16:
Hey Peter,

You probably know this and probably have a better way of handling it.  But here's how I'm handling it for now.

populateMacroButtons is run whenever delete or apply is clicked in the Macro popup.  populateMacroButtons calls executeJS so that it will run at startup.  I've added a simple argument to populateMacroButtons  in order to know if its being called during startup or because of a macro change.

```
function populateMacroButtons(firstRun = 0) {

...

if (firstRun != 0) {
  printLogModern(icon, source, string, printLogCls)
  executeJS(buttonsarray[i].javascript)
}

...
```

Then in the call when app starts

```
$(document).ready(function() {
  populateMacroButtons(1)
  bindKeys()
});
```

petervanderwalt, 2021-03-16:
Your solution is perfect, it flags the first run and only the first run. Well done. Implementing basically as shown (:

closed 2021-03-16 by petervanderwalt