OpenBuilds/OpenBuilds-CONTROL #176: Problem when maximizing window on KDE

federicort, 2021-01-12:
I have Kubuntu 20.04, I'm running Openbuilds Control 1.0.273, and when I try to maximize the window, the program freezes, and I can't get it to work again until I close it and open it again

petervanderwalt, 2021-01-13:
Try running it from the CLI with ELECTRONDEBUG=true ./<exetuble> to get logging turned on - see what Electron says when it maximises.  CONTROL just uses the standard Electron APIs for minimize, maximize,etc: https://www.electronjs.org/docs/api/browser-window#winmaximize

closed 2021-01-20 by petervanderwalt

federicort, 2021-01-22:
Sorry for the delay, so I went to the folder where the executable is, run `./openbuildscontrol` in the terminal.

As soon as it opens it gives me this:

> (node:8312) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
[Error: EACCES: permission denied, open '/opt/OpenBuildsCONTROL/resources/app.asar.unpacked/firmware.bin'] {
  errno: -13,
  code: 'EACCES',
  syscall: 'open',
  path: '/opt/OpenBuildsCONTROL/resources/app.asar.unpacked/firmware.bin'
}
Error [ERR_STREAM_DESTROYED]: Cannot call write after a stream was destroyed
    at doWrite (_stream_writable.js:431:19)
    at writeOrBuffer (_stream_writable.js:419:5)
    at WriteStream.Writable.write (_stream_writable.js:309:11)
    at IncomingMessage.ondata (_stream_readable.js:728:22)
    at IncomingMessage.emit (events.js:223:5)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:290:11)
    at IncomingMessage.Readable.push (_stream_readable.js:224:10)
    at HTTPParser.parserOnBody (_http_common.js:128:24)
    at TLSSocket.socketOnData (_http_client.js:456:22) {
  code: 'ERR_STREAM_DESTROYED'
}
Error: Parse Error: JS Exception
    at TLSSocket.socketOnData (_http_client.js:456:22)
    at TLSSocket.emit (events.js:223:5)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:290:11)
    at TLSSocket.Readable.push (_stream_readable.js:224:10)
    at TLSWrap.onStreamRead (internal/stream_base_commons.js:181:23) {
  bytesParsed: -6302,
  code: 'HPE_JS_EXCEPTION',
  reason: 'JS Exception',
  rawPacket: <Buffer 15 40 31 1b 15 40 31 1b 15 40 31 1b 15 40 88 1e 15 40 31 1b 15 40 25 1b 15 40 16 1b 15 40 88 1e 15 40 88 1e 15 40 88 1e 15 40 31 1b 15 40 16 1b 15 40 ... 1328 more bytes>
}

and moments later shows this:

> Checking for update
Error: Error: ENOENT: no such file or directory, open '/opt/OpenBuildsCONTROL/resources/app-update.yml'
(node:8312) UnhandledPromiseRejectionWarning: Error: ENOENT: no such file or directory, open '/opt/OpenBuildsCONTROL/resources/app-update.yml'
(node:8312) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:8312) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

but nothing after I maximized the window and it freezes.