OpenBuilds/OpenBuilds-CONTROL #177: v1.0.274 does not launch on Ubuntu 20.04.1 LTS: TLS error?

swetland, 2021-01-15:
I see this with both the appimage and the .deb version, and in 1.0.273 as well:

$ **./OpenBuildsCONTROL-1.0.274.AppImage** 
(node:1336277) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
[Error: EROFS: read-only file system, open '/tmp/.mount_OpenBuuKpZIa/resources/app.asar.unpacked/firmware.bin'] {
  errno: -30,
  code: 'EROFS',
  syscall: 'open',
  path: '/tmp/.mount_OpenBuuKpZIa/resources/app.asar.unpacked/firmware.bin'
}
Error [ERR_STREAM_DESTROYED]: Cannot call write after a stream was destroyed
    at doWrite (_stream_writable.js:431:19)
    at writeOrBuffer (_stream_writable.js:419:5)
    at WriteStream.Writable.write (_stream_writable.js:309:11)
    at IncomingMessage.ondata (_stream_readable.js:728:22)
    at IncomingMessage.emit (events.js:223:5)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:290:11)
    at IncomingMessage.Readable.push (_stream_readable.js:224:10)
    at HTTPParser.parserOnBody (_http_common.js:128:24)
    at TLSSocket.socketOnData (_http_client.js:456:22) {
  code: 'ERR_STREAM_DESTROYED'
}
Error: Parse Error: JS Exception
    at TLSSocket.socketOnData (_http_client.js:456:22)
    at TLSSocket.emit (events.js:223:5)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:290:11)
    at TLSSocket.Readable.push (_stream_readable.js:224:10)
    at TLSWrap.onStreamRead (internal/stream_base_commons.js:181:23) {
  bytesParsed: 2914,
  code: 'HPE_JS_EXCEPTION',
  reason: 'JS Exception',
  rawPacket: <Buffer 14 40 4d f8 14 40 4d f8 14 40 4d f8 14 40 a4 fb 14 40 4d f8 14 40 41 f8 14 40 32 f8 14 40 a4 fb 14 40 a4 fb 14 40 a4 fb 14 40 4d f8 14 40 32 f8 14 40 ... 1328 more bytes>
}

At this point it hangs forever.

heschlie, 2021-01-19:
Seeing the same thing on Arch no extra output from launching with `ELECTRONDEBUG=true`, the application launches in the tray but upon opening the GUI it just shows a grey box. It is also putting a pretty heavy load on the CPU the whole time. I picked a randomly old version around the time I remember this working which was `1.0.199` and it worked. I did try every version back to 1.0.253` before jumping way back and they all showed the same issue.

OS: Linux Arch
DE: KDE Plasma

petervanderwalt, 2021-01-20:
Merging with bug #179

closed 2021-01-20 by petervanderwalt

petervanderwalt, 2021-02-01:
please test CONTROL v 1.0.278 or later, see if that resolves the issue (does for Windows, but I do not have an Arch or Ubuntu set-up at hand at this very moment) 

https://github.com/OpenBuilds/OpenBuilds-CONTROL/releases/tag/v1.0.278