OpenBuilds/OpenBuilds-CONTROL #183: Can we have a display of the actual feed rate?

JustinTimeToo, 2021-01-28:
I'd like to have a display of the actual feed rate and see it change when I move the feed slider to show the current feed rate (not in percentage)

petervanderwalt, 2021-02-01:
Added in v1.0.277 

[image]

sharmstr, 2021-02-03:
Hey Peter, this doesnt appear to be working.  I have a 2 hour job running right now so cant test, but I noticed it said 0mm/min at 100% so I changed it to 105% but it only shows the percentage difference(?).  I'm on .282

[image]

sharmstr, 2021-02-03:
Just went back to the machine and even at 140% it shows 0mm/min with a blip up to 1mm/min every couple of seconds.

petervanderwalt, 2021-02-04:
Thanks, will check!

petervanderwalt, 2021-02-11:
Fixed in 1.0.283

closed 2021-02-11 by petervanderwalt

sharmstr, 2021-02-14:
Hey Peter.  This appears to be fixed and it working great.  Thank you.