OpenBuilds/OpenBuilds-CONTROL #187: V1.0.280: Incremental Jog 0.1mm is something larger than 0.1mm

brrian11, 2021-02-02:
mm-mode, Incremental jog, 0.1mm: if I had to guess, it's jogging .0001" instead of 0.1mm.  One at a time from zero, the DRO reads:

.03mm
.05mm
.08mm
.1mm

...and so on.

sharmstr, 2021-02-02:
Peter, this happens when switching from inch mode to mm mode..  The jog distance will stay at .0254.  If I jog 1mm, 10mm or 100mm then switch back to .1mm, it works as expected.

I had the same issue when I wrote my macro code.  Let me dig that up and see how I fixed it.

sharmstr, 2021-02-02:
I believe its in your mmMode function.  There's no if for .0254 so it wont change to .1mm

`function mmMode() {
  unit = "mm";
  localStorage.setItem('unitsMode', unit);
  $('#dist01label').html('0.1mm')
  $('#dist1label').html('1mm')
  $('#dist10label').html('10mm')
  $('#dist100label').html('100mm')
  if (jogdist == 0.254) {
    jogdist = 0.1
  }
  if (jogdist == 2.54) {
    jogdist = 1
  }
  if (jogdist == 25.4) {
    jogdist = 10
  }
  if (jogdist == 254) {
    jogdist = 100
  }
  $('#jogratemmdiv').show()
  $('#jograteinchdiv').hide()
  inchtommrate();
}`

I injected this into .280 and it seems to have fixed it.

`window.mmMode = function() {
    unit = "mm";
    localStorage.setItem('unitsMode', unit);
    $('#dist01label').html('0.1mm')
    $('#dist1label').html('1mm')
    $('#dist10label').html('10mm')
    $('#dist100label').html('100mm')
    if (jogdist == 0.0254) {
      jogdist = 0.1
    }
    if (jogdist == .254) {
      jogdist = 1
    }
    if (jogdist == 2.54) {
      jogdist = 10
    }
    if (jogdist == 25.4) {
      jogdist = 100
    }
    $('#jogratemmdiv').show()
    $('#jograteinchdiv').hide()
    inchtommrate();
  }`

petervanderwalt, 2021-02-02:
On it!

petervanderwalt, 2021-02-02:
Thanks for doing the Debug @sharmstr ! - saved me some time, so here we go, v1.0.281 will be up shortly
Fixed in https://github.com/OpenBuilds/OpenBuilds-CONTROL/commit/55a0c077753d4d67a3d0fc634b0a0e82c480b807

petervanderwalt, 2021-02-02:
(; even managed to sneak a new feature in in the same release (; https://github.com/OpenBuilds/OpenBuilds-CONTROL/commit/55a0c077753d4d67a3d0fc634b0a0e82c480b807#diff-ff905774288e5cd43c5adcde46191c8990ae3fc1472fe5315344493aa946aadeR1

sharmstr, 2021-02-18:
Hey Peter,

Similar issue with changing step size via keyboard.  There's no catch for .0254.  Here's the code I used that seems to work ok.

`''
window.changeStepSize = function(dir) {
  if (jogdist == 0.1 || jogdist == .0254) {
      if (dir == 1) {
        if (unit == "mm") {
          jogdist = 1;
        } else if (unit == "in") {
          jogdist = .254;
        }
        $('.distbtn').removeClass('bd-openbuilds')
        $('#dist1').addClass('bd-openbuilds')
        $('.jogdist').removeClass('fg-openbuilds')
        $('.jogdist').addClass('fg-gray')
        $('#dist1label').removeClass('fg-gray')
        $('#dist1label').addClass('fg-openbuilds')
      }
      if (dir == -1) {
        // do nothing
      }
    } else if (jogdist == 1 || jogdist == .254) {
      if (dir == 1) {
        if (unit == "mm") {
          jogdist = 10;
        } else if (unit == "in") {
          jogdist = 2.54;
        }
        $('.distbtn').removeClass('bd-openbuilds')
        $('#dist10').addClass('bd-openbuilds')
        $('.jogdist').removeClass('fg-openbuilds')
        $('.jogdist').addClass('fg-gray')
        $('#dist10label').removeClass('fg-gray')
        $('#dist10label').addClass('fg-openbuilds')
      }
      if (dir == -1) {
        if (unit == "mm") {
          jogdist = .01;
        } else if (unit == "in") {
          jogdist = .0254;
        }
        $('.distbtn').removeClass('bd-openbuilds')
        $('#dist01').addClass('bd-openbuilds')
        $('.jogdist').removeClass('fg-openbuilds')
        $('.jogdist').addClass('fg-gray')
        $('#dist01label').removeClass('fg-gray')
        $('#dist01label').addClass('fg-openbuilds')
      }
    } else if (jogdist == 10 || jogdist == 2.54) {
      if (dir == 1) {
        if (unit == "mm") {
          jogdist = 100;
        } else if (unit == "in") {
          jogdist = 25.4;
        }
        $('.distbtn').removeClass('bd-openbuilds')
        $('#dist100').addClass('bd-openbuilds')
        $('.jogdist').removeClass('fg-openbuilds')
        $('.jogdist').addClass('fg-gray')
        $('#dist100label').removeClass('fg-gray')
        $('#dist100label').addClass('fg-openbuilds')
      }
      if (dir == -1) {
        if (unit == "mm") {
          jogdist = 1;
        } else if (unit == "in") {
          jogdist = .254;
        }
        $('.distbtn').removeClass('bd-openbuilds')
        $('#dist1').addClass('bd-openbuilds')
        $('.jogdist').removeClass('fg-openbuilds')
        $('.jogdist').addClass('fg-gray')
        $('#dist1label').removeClass('fg-gray')
        $('#dist1label').addClass('fg-openbuilds')
      }
    } else if (jogdist == 100 || jogdist == 25.4) {
      if (dir == 1) {
        // do nothing
      }
      if (dir == -1) {
        if (unit == "mm") {
          jogdist = 10;
        } else if (unit == "in") {
          jogdist = 2.54;
        }
        $('.distbtn').removeClass('bd-openbuilds')
        $('#dist10').addClass('bd-openbuilds')
        $('.jogdist').removeClass('fg-openbuilds')
        $('.jogdist').addClass('fg-gray')
        $('#dist10label').removeClass('fg-gray')
        $('#dist10label').addClass('fg-openbuilds')
      }
    }
  }
''`

petervanderwalt, 2021-02-26:
Hopefully all fixed in latest version (:

sharmstr, 2021-03-02:
Hmmm.  Still not working correctly.  But now its different.  When in inch mode, switching step via keyboard sets it to mm value (.01,.1,10,100) instead of inch.  Also gets stuck on .1mm/.001" button.  Will have a look at it tomorrow morning.

sharmstr, 2021-03-02:
Okay, its my code that's making the button stick.  So forget that.

But I just looked at your changes and changing the if jogdist statements isnt enough.  You also need to check if what unit you are in and set the appropriate value.  Example

```
if (unit == "mm") {
  jogdist = 1;
} else if (unit == "in") {
  jogdist = .254;
 }
```

petervanderwalt, 2021-03-10:
Good catch! I didn't see that - it mustve been broken since forever and never worked with inch mode! Yikes 

Fixed (and tested) in v1.0.291 and up (wasn't near controller to test with on last fix sorry) 

Many thanks for helping debug and test!

closed 2021-03-10 by petervanderwalt