OpenBuilds/OpenBuilds-CONTROL #189: Issue with build 280

HalfFinn52, 2021-02-02:
After upgrading to 280 on my Raspberry Pi, when the software connects to the Arduino, although it identifies the GRBL software, it doesn't seem to complete the steps necessary and there is no alarm set (that I can clear) and the Home button is greyed out.  On my Windows machine, I was able to go back to an earlier release that I had downloaded.  Due to the nature of the Raspberry Pi install, I am unable to do that.

Thanks,

petervanderwalt, 2021-02-02:
Custom Grbl Fork?
Any errors in devtools Console?

HalfFinn52, 2021-02-02:
[image]

The GRBL version is 1.1h, which I don't believe is any special fork.

Sorry, but I am not sure how to view the devtools Console to see if there are any errors.

petervanderwalt, 2021-02-02:
In the browser window, press Ctrl+Shift+i to open Devtools, then click the Console Tab https://developers.google.com/web/tools/chrome-devtools

HalfFinn52, 2021-02-02:
Hi Peter, I uploaded the HEX file on the OpenBuilds support forum.  With a new install of 281 on the Raspberry Pi machine, when I went to the devtools Console, I got the info in the attached screen capture.
[image]

petervanderwalt, 2021-02-02:
Same person as https://openbuilds.com/threads/v1-0-280-issues.17298/#post-110307? (:   

Already replied on Forum then, lets consider this thread a "duplicate" of the ongoing forum discussion.

closed 2021-02-02 by petervanderwalt

HalfFinn52, 2021-02-02:
Agreed.