OpenBuilds/OpenBuilds-CONTROL #195: Job completion dialog pops up on Z+ jog using keyboard shortcut

evilc66, 2021-02-24:
After getting in a bluetooth number pad to use as a jog pendant (totally awesome btw once you put custom labels on the keys), I have found that every time I press Z+ (page up) in incremental or continuous jog modes, the "job completion" dialog pops up with no time elapsed. Does the same on my primary keyboard. The GUI Z+ jog doesn't have the same issue.

sharmstr, 2021-03-02:
I was able to reproduce this.  Check your keyboard shortcut settings.  I'm willing to bet that you have Run/Pause assigned to pageup as well as Z+.

Hey Peter,  I thought at one point Control looked for duplicate key assignments.  Maybe I'm thinking of macro key assignments.

petervanderwalt, 2021-03-10:
Agreed, may be related to customized keys, default does not cause the issue. 

We don't check for duplicates (but i guess we should)

petervanderwalt, 2021-03-10:
v1.0.291 and up will check for duplicates 

[image]

closed 2021-03-10 by petervanderwalt

evilc66, 2021-03-10:
Sorry for taking a while to get back to you guys.

Turns out I did have a duplicate key assigned to Run/Pause somehow. All fixed now.

petervanderwalt, 2021-03-10:
Did away with the red toast error, now showing error right below key you are trying to assign: 
(also crosschecks internal key assignments, macros, etc to really cut down on accidental duplicates)

[image]

Should be out in a week or so