OpenBuilds/OpenBuilds-CONTROL #200: Auto updates

petervanderwalt, 2021-03-17:
https://openbuilds.com/threads/cruising-along-and-bam-it-just-stops.17584/

petervanderwalt, 2021-03-17:
Todo: Fix GH Actions release script to prepend "v" before tag

sharmstr, 2021-03-21:
For what its worth, I got this error on my machine running .292.  My desktop didnt have this problem.

Error in auto-updater: Cannot check for updates: Error: Cannot find latest.yml in the latest release artifacts (https://github.com/OpenBuilds/OpenBuilds-CONTROL/releases/download/v1.0.294/latest.yml): HttpError: 404 "method: GET url: https://github.com/OpenBuilds/OpenBuilds-CONTROL/releases/download/v1.0.294/latest.yml\n\nPlease double check that your authentication token is correct. Due to security reasons actual status maybe not reported, but 404.\n" Headers: { "status": "404", "server": "GitHub.com", "date": "Sun, 21 Mar 2021 16:04:56 GMT", "content-type": "text/plain; charset=utf-8", "vary": "X-PJAX, Accept-Encoding, Accept, X-Requested-With", "cache-control": "no-cache", "strict-transport-security": "max-age=31536000; includeSubdomains; preload", "x-frame-options": "deny", "x-content-type-options": "nosniff", "x-xss-protection": "0", "referrer-policy": "no-referrer-when-downgrade", "expect-ct": "max-age=2592000, report-uri=\"https://api.github.com/_private/browser/errors\"", "content-security-policy": "default-src 'none'; base-uri 'self'; connect-src 'self'; form-action 'self'; img-src 'self' data:; script-src 'self'; style-src 'unsafe-inline'", "content-encoding": "gzip", "content-length": "29", "x-github-request-id": "C25F:0502:E89425:100E8A4:60576EA8" } at createHttpError (C:\Program Files\OpenBuildsCONTROL\resources\app.asar\node_modules\builder-util-runtime\out\httpExecutor.js:84:10) at ElectronHttpExecutor.handleResponse (C:\Program Files\OpenBuildsCONTROL\resources\app.asar\node_modules\builder-util-runtime\out\httpExecutor.js:169:14) at ClientRequest. (C:\Program Files\OpenBuildsCONTROL\resources\app.asar\node_modules\builder-util-runtime\out\httpExecutor.js:134:16) at ClientRequest.emit (events.js:223:5) at SimpleURLLoaderWrapper. (electron/js2c/browser_init.js:93:6152) at SimpleURLLoaderWrapper.emit (events.js:223:5) at newError (C:\Program Files\OpenBuildsCONTROL\resources\app.asar\node_modules\builder-util-runtime\out\index.js:212:17) at GitHubProvider.getLatestVersion (C:\Program Files\OpenBuildsCONTROL\resources\app.asar\node_modules\electron-updater\out\providers\GitHubProvider.js:135:50) at async NsisUpdater.getUpdateInfoAndProvider (C:\Program Files\OpenBuildsCONTROL\resources\app.asar\node_modules\electron-updater\out\AppUpdater.js:500:13) at async NsisUpdater.doCheckForUpdates (C:\Program Files\OpenBuildsCONTROL\resources\app.asar\node_modules\electron-updater\out\AppUpdater.js:516:20)

petervanderwalt, 2021-03-21:
Please hit F5 and see if it sitll does? (renamed tag and release under releases)

sharmstr, 2021-03-21:
All good!

nirkki, 2021-03-24:
Autoupdate fails with v1.0.295 as well:

`Error: Cannot find latest.yml in the latest release artifacts (https://github.com/OpenBuilds/OpenBuilds-CONTROL/releases/download/1.0.295/latest.yml)
`

 There is no "download" directory under "releases" in GitHub? Mayby "tag" instead of "download"?

sharmstr, 2021-03-24:
Did you try hitting F5 in Control?

nirkki, 2021-03-24:
> Did you try hitting F5 in Control?
> 

No effect, still the same:

```
[16:08:21] [ update ] Checking for Updates
[16:08:21] [ Update Available! ] OpenBuilds CONTROL 1.0.295. is available now.
[16:08:21] [ Update Available! ] Download will start in 0 seconds (cancel)
[16:08:21] [ Update Available! ] You will be prompted when its ready to be installed
[16:08:32] [ autoupdate ] Starting update... Please wait
[16:08:33] [ autoupdate ] Error in auto-updater: Cannot check for updates: Error: Cannot find latest.yml in the latest release artifacts (https://github.com/OpenBuilds/OpenBuilds-CONTROL/releases/download/1.0.295/latest.yml): HttpError: 404 "method: GET url: https://github.com/OpenBuilds/OpenBuilds-CONTROL/releases/download/1.0.295/latest.yml
etc
```

nirkki, 2021-03-24:
I think it's the missing "v" ?
v1.0.294 tries to load the update from 

> https://github.com/OpenBuilds/OpenBuilds-CONTROL/releases/download/1.0.295/latest.yml

instead of

> https://github.com/OpenBuilds/OpenBuilds-CONTROL/releases/download/v1.0.295/latest.yml

petervanderwalt, 2021-03-25:
> I think it's the missing "v" ?

Correct, as mentioned above: 

> Todo: Fix GH Actions release script to prepend "v" before tag

However there is also an issue upstream from us in that the JSON for the latest release DOES include the v (see https://api.github.com/repos/OpenBuilds/OpenBuilds-CONTROL/releases/latest) but not always... We are working with Github https://github.com/OpenBuilds/OpenBuilds-CONTROL/actions but it may also be a bug in https://www.npmjs.com/package/electron-updater

To clarify, both electron-updated and github-actions are external providers, if it was in our code we could fix it easier

petervanderwalt, 2021-04-01:
Looking at the source of Electron-updater: https://github.com/electron-userland/electron-builder/blob/master/packages/electron-updater/src/providers/GitHubProvider.ts#L130-L132

I may just need to set the vPrefixedTagName option to true in package.json :D

[image]

Yip, that works, vPrefixedTagName : true sorts out the update.
Only catch is a couple different scenarios out in the field:

People running versions looking for v1.0.0 (old versions, like before 1.0.284, and newer versions after 1.09.297)
People running versions looking for 1.0.0 (like yours, probably between 1.0.285 and 1.0.296)

And both look at "latest" - so for next couple days I'll rename the release on alternate days :) so both gets a chance to grab 1.0.297 from whenceforth they'll be good with us naming it with "v"

closed 2021-04-16 by petervanderwalt