OpenBuilds/OpenBuilds-CONTROL #210: Job complete time and filename bugs

petervanderwalt, 2021-04-10:
@sharmstr reported two good ones https://openbuilds.com/threads/job-completed-popup-missing-in-1-0-298.17732/#post-114171

petervanderwalt, 2021-04-16:
Path issue from files posted via the API: Fixed backend to stop including the full path
```
var filename = path.parse(filePath)
filename = filename.name + filename.ext
```
Also limited column width, thanks for the idea

petervanderwalt, 2021-04-16:
Job timer:  Upon completing job, update to 

[image]

sharmstr, 2021-04-16:
I guess what confused me about the first time stamp and what it was for is because mine isnt updating while the job runs.

Also, just installed 299.  The job log isnt formatted and first time stamp doesnt say "done".  I did stop the job early.

[image]

petervanderwalt, 2021-04-16:
I think there may be a Build issue, I downloaded the build, and indeed the changes are not included.  Incremented build number – pushing again, will download and test v1.0.300 before releasing

From: sharmstr
Sent: Friday, 16 April 2021 16:02
To: OpenBuilds/OpenBuilds-CONTROL
Cc: Peter van der Walt; State change
Subject: Re: [OpenBuilds/OpenBuilds-CONTROL] Job complete time and filenamebugs (#210)

I guess what confused me about the first time stamp and what it was for is because mine isnt updating while the job runs.
Also, just installed 299. The job log isnt formatted and first time stamp doesnt say "done". I did stop the job early.

sharmstr, 2021-04-16:
Getting closer.  The job log format is awesome.  Thank you.

But the first timestamp is still whacky

[image]

petervanderwalt, 2021-04-16:
> EDIT: Looks like this only happens when you stop a job before its done.

Could be! I do a cleanup when a job is complete.

closed 2021-05-26 by petervanderwalt