OpenBuilds/OpenBuilds-CONTROL #211: DRO not updating properly using Shift+Enter

brrian11, 2021-04-24:
This happened again in 1.0.300

After I ran a program, using Shift+Enter I tried changing Z to 1.695in and got the result 43.053in.

Same as: [DRO not updating properly using Shift+Enter](https://github.com/OpenBuilds/OpenBuilds-CONTROL/issues/202)

sharmstr, 2021-04-27:
@petervanderwalt Looks like you forgot to add the G21 to y and z.

For X your have

sendGcode("**G21**\nG10 P0 L20 X" + $("#xPosInput").val());

But Y and Z doesnt have it.  I tested and it seems like X works fine.  @brrian11 can you confirm?

brrian11, 2021-04-28:
Sorry for the late reply. I did this again, changed Z first & got an incorrect number in the DRO. Then I changed X and it worked correctly, then Y and it worked correctly. Then I changed Z again and it also worked correctly. @sharmstr - if you're correct, when I changed X it 'set the correct mode', then everything after that worked. I'm running a long program but will check afterwards - I'll change Z, then Y, then X & see what happens.

sharmstr, 2021-04-28:
Yep.  X issues the G21 so everything after that will work.  Quit test is to issue G20 in the serial console to switch it back to inch mode.

brrian11, 2021-04-29:
& yeah, I just changed Z (=wrong), then Y (=wrong), then X (=right), then Y (=right), then Z (=right). So, it's what you think. Thanks...

closed 2021-05-11 by petervanderwalt